espressif/esp-protocols #636: [websocket] ability to set TCP_NODELAY

bryghtlabs-richard, 2024-08-19:
### Is your feature request related to a problem?

Our customers play a turn-based game, sometimes their moves are delayed by ~40ms. Not every time, but often enough that it's a small burden in some of the quicker time formats.

### Describe the solution you'd like.

I'd like the ability to dynamically configure TCP_NODELAY, or instead the ability to call setsockopt() myself at runtime. It should be runtime-configurable, as outside of games we send lots of websocket frame that can benefit from Nagling, but during quick games I need moves to go out on schedule.

### Describe alternatives you've considered.

I've considered switching to UDP, but I'd have to implement reliability and connection myself, and websockets are a great fit for what we need here.

bryghtlabs-richard, 2024-08-19:
Looks like this may need to be wired into the transport layer(ssl vs tcp) too

euripedesrocha, 2024-08-20:
Hi @bryghtlabs-richard thanks for the request. 
You are right, we need to set it in the transport level. We are already introducing the possibility to tcp_transport and will made it available on websocket after that. 

The workaround to have it before we finish our process is to create a custom transport based on the ones available and use it on the websocket client.

bryghtlabs-richard, 2024-12-02:
I ended up rewriting my transport_ws.c to write-combine headers and payload, so this is a low-priority request from us, as the increased latency only impacts multiple WS frames, which is uncommon here.

bryghtlabs-richard, 2024-12-02:
I added a workaround for espressif/esp-idf#14918, and when I did it improved network latency, even without TCP_NODELAY. Because WebSocket writes are being split over multiple TCP packets, the first packet goes out immediately, but then Nagle's algorithm delays the second packet until the server ACKs the first packet. Adding write-combining to transport_ws.c helps this out because short WS frames are sent as single TCP packets. That said, I look forward to TCP_NODELAY being available for esp-websocket-client, because it still introduces extra latency in the case where two short WS frames are sent back to back as single TCP packets.

I noticed this has "status:done", but I can't seem to find anything related in esp-protocols. Is this on the way? Thank you for your help with this.

euripedesrocha, 2024-12-03:
@bryghtlabs-richard the approach we went here was to "open" the transport layer, giving user access to the socket. 

So you can get the socket and apply the settings there. I would recommend doing it from the before connect event when convenient. Unfortunately `esp_transport` still lacks documentation, it was always considered an internal component, once it is done we would add the details and considerations. Let me know if you have any questions.

closed 2024-12-03 by espressif-bot

bryghtlabs-richard, 2024-12-03:
Thanks @euripedesrocha ! That works nicely.