espressif/esp-protocols #637: esp-netif_lwip-ppp: Connection lost, without event IP_EVENT_PPP_GOT_IP

ezamp, 2024-08-20:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm getting _esp-netif_lwip-ppp: Connection lost_ on SIM7028 module. From the logs, the module responds to the AT commands but when I switch it to the data mode I can't get IP and after a while it displays _IP event 7_ (IP_EVENT_PPP_LOST_IP). 
The APN and other configurations seem correct, and on the sim provider's portal I see that the PDP context is created correctly.
[image]

Log:
```
I (26) boot: ESP-IDF v5.3-dirty 2nd stage bootloader
I (27) boot: compile time Aug 20 2024 13:52:01
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=129ech ( 76268) map
I (113) esp_image: segment 1: paddr=00022a14 vaddr=3fc91d00 size=028f0h ( 10480) load
I (116) esp_image: segment 2: paddr=0002530c vaddr=40374000 size=0ad0ch ( 44300) load
I (129) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=3a470h (238704) map
I (172) esp_image: segment 4: paddr=0006a498 vaddr=4037ed0c size=02fd4h ( 12244) load
I (181) boot: Loaded app from partition at offset 0x10000
I (181) boot: Disabling RNG early entropy source...
I (193) cpu_start: Multicore app
D (193) cpu_start: Pro cpu up
D (193) cpu_start: Starting app cpu, entry point is 0x40375198
0x40375198: call_start_cpu1 at /Users/emanuele/esp/v5.3/esp-idf/components/esp_system/port/cpu_start.c:195

D (0) cpu_start: App cpu up
D (203) clk: RTC_SLOW_CLK calibration value: 3543539
D (214) cpu_start: calling init function: 0x42002798 on core: 0
0x42002798: __esp_system_init_fn_init_efuse_check at /Users/emanuele/esp/v5.3/esp-idf/components/efuse/src/esp_efuse_startup.c:31

D (214) cpu_start: calling init function: 0x420028a0 on core: 0
0x420028a0: __esp_system_init_fn_init_show_cpu_freq at /Users/emanuele/esp/v5.3/esp-idf/components/esp_system/startup_funcs.c:58

I (215) cpu_start: Pro cpu start user code
I (219) cpu_start: cpu freq: 160000000 Hz
D (224) cpu_start: calling init function: 0x420026b8 on core: 0
0x420026b8: __esp_system_init_fn_init_show_app_info at /Users/emanuele/esp/v5.3/esp-idf/components/esp_app_format/esp_app_desc.c:104

I (230) app_init: Application information:
I (235) app_init: Project name:     nbTest
I (240) app_init: App version:      1
I (244) app_init: Compile time:     Aug 20 2024 13:51:59
I (250) app_init: ELF file SHA256:  00850e25c...
I (255) app_init: ESP-IDF:          v5.3-dirty
D (260) cpu_start: calling init function: 0x420027ac on core: 0
0x420027ac: __esp_system_init_fn_init_efuse_show_app_info at /Users/emanuele/esp/v5.3/esp-idf/components/efuse/src/esp_efuse_startup.c:41

I (266) efuse_init: Min chip rev:     v0.0
I (271) efuse_init: Max chip rev:     v0.99 
I (276) efuse_init: Chip rev:         v0.2
D (281) cpu_start: calling init function: 0x42004144 on core: 0
0x42004144: __esp_system_init_fn_init_heap at /Users/emanuele/esp/v5.3/esp-idf/components/heap/heap_caps_init.c:25

D (287) memory_layout: Checking 6 reserved memory ranges:
D (292) memory_layout: Reserved memory range 0x3fc84000 - 0x3fc91d00
[... 280 lines cut ...]
        return;
    }
    ESP_LOGI(TAG, "Signal quality: rssi=%d, ber=%d", rssi, ber);
    
    char imsi[100];
    err = esp_modem_get_imsi(dce, &imsi);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "esp_modem_get_imsi failed with %d %s", err, esp_err_to_name(err));
        return;
    }
    ESP_LOGI(TAG, "imsi=%s", imsi);

    err = esp_modem_set_mode(dce, ESP_MODEM_MODE_DATA);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "esp_modem_set_mode(ESP_MODEM_MODE_DATA) failed with %d", err);
        return;
    }

    // Wait for IP address 
    ESP_LOGI(TAG, "Waiting for IP address");
    xEventGroupWaitBits(event_group, CONNECT_BIT, pdFALSE, pdFALSE, portMAX_DELAY);
}
```

Thanks

ezamp, 2024-08-20:
Hi @david-cermak

Thanks for your reply, I enabled _CONFIG_LWIP_PPP_DEBUG_ON_ but I don't see difference in the log. 
Now the log level is configured to DEBUG.

Log: 
```
[...]
I (683) test: End setup
D (693) esp_netif_lwip: check: remote, if=0x3fc98364 fn=0x42013158
0x42013158: esp_netif_new_api at /Users/emanuele/esp/v5.3/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:690

D (693) esp_netif_objects: esp_netif_add_to_list_unsafe netif added successfully (total netifs: 1)
D (703) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3fc9bfc0
D (713) esp-netif_lwip-ppp: Phase Dead
D (713) esp_netif_lwip: call api in lwip: ret=0x0, give sem
I (723) test: Initializing esp_modem for the NB-IoT module...
I (723) uart: queue free spaces: 30
D (733) intr_alloc: Connected src 28 to int 8 (cpu 0)
I (743) test: Signal quality: rssi=22, ber=0
I (753) test: imsi=901405108749214
D (753) command_lib: generic_command command ATE0

D (753) command_lib: Response: 
OK

D (753) command_lib: generic_command command AT+CGDCONT=1,"IP","iot.1nce.net"

D (953) command_lib: Response: 
OK

D (963) esp_netif_handlers: esp_netif action has started with netif0x3fc9bd8c from event_id=0
D (973) esp_netif_lwip: check: remote, if=0x3fc9bd8c fn=0x420133ec
0x420133ec: esp_netif_start_api at /Users/emanuele/esp/v5.3/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1068

D (973) esp_netif_lwip: esp_netif_start_api 0x3fc9bd8c
D (983) esp-netif_lwip-ppp: esp_netif_start_ppp: Starting PPP connection: 0x3fc9bfc0
D (993) esp-netif_lwip-ppp: Phase Start
D (993) esp-netif_lwip-ppp: Phase Establish
D (993) esp_netif_lwip: check: local, if=0x3fc9bd8c fn=0x420138cc
0x420138cc: esp_netif_update_default_netif_lwip at /Users/emanuele/esp/v5.3/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:317

D (1003) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3fc9bd8c
D (1013) esp_netif_lwip: call api in lwip: ret=0x0, give sem
I (1013) test: Waiting for IP address
D (60993) esp-netif_lwip-ppp: Phase Disconnect
D (60993) esp-netif_lwip-ppp: Phase Dead
I (60993) esp-netif_lwip-ppp: Connection lost
D (60993) event: running post IP_EVENT:7 with handler 0x4200aa2c and context 0x3fc9bd38 on loop 0x3fc9adec
0x4200aa2c: on_ip_event at /Users/emanuele/Documents/Projects/nbTest/main/main.c:29

I (61003) test: IP event! 7
I (61003) test: Modem Disconnect from PPP Server
D (61013) event: running post IP_EVENT:7 with handler 0x42012614 and context 0x3fca0818 on loop 0x3fc9adec
0x42012614: esp_netif_action_disconnected at /Users/emanuele/esp/v5.3/esp-idf/components/esp_netif/esp_netif_handlers.c:83

D (61023) esp_netif_handlers: esp_netif action disconnected with netif0x3fc9bd8c from event_id=7
D (61033) esp_netif_lwip: check: remote, if=0x3fc9bd8c fn=0x42012eb0
0x42012eb0: esp_netif_down_api at /Users/emanuele/esp/v5.3/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1673

D (61033) esp_netif_lwip: esp_netif_down_api esp_netif:0x3fc9bd8c
D (61043) esp_netif_lwip: esp_netif_internal_dhcpc_cb lwip-netif:0x3fc9be14
D (61043) esp_netif_lwip: check: local, if=0x3fc9bd8c fn=0x420138cc
0x420138cc: esp_netif_update_default_netif_lwip at /Users/emanuele/esp/v5.3/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:317

D (61053) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3fc9bd8c
D (61063) esp_netif_lwip: call api in lwip: ret=0x0, give sem
```

david-cermak, 2024-08-20:
oh, sorry, forgot about the `CONFIG_LWIP_DEBUG`, this needs to be enabled, too, to propagate the logs from the liwp.

ezamp, 2024-08-20:
Okay, now I see the logs
```
I (5760) test: Signal quality: rssi=99, ber=99
I (5770) test: imsi=901405108749214
D (5770) command_lib: generic_command command ATE0

D (5780) command_lib: Response: 
OK

D (5780) command_lib: generic_command command AT+CGDCONT=1,"IP","iot.1nce.net"

D (5870) command_lib: Response: 
OK

D (5870) command_lib: generic_command command ATD*99#

D (5870) command_lib: Response: 
ERROR

D (5870) command_lib: generic_command command ATO

D (5880) command_lib: Response: 
ERROR

D (5880) esp_netif_handlers: esp_netif action has started with netif0x3fc9bd8c from event_id=0
D (5890) esp_netif_lwip: check: remote, if=0x3fc9bd8c fn=0x42013684
0x42013684: esp_netif_start_api at /Users/emanuele/esp/v5.3/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1068

D (5900) esp_netif_lwip: esp_netif_start_api 0x3fc9bd8c
D (5900) esp-netif_lwip-ppp: esp_netif_start_ppp: Starting PPP connection: 0x3fc9bfc0
ppp_connect[1]: holdoff=0
ppp phase changed[1]: phase=3
D (5910) esp-netif_lwip-ppp: Phase Start
pppos_connect: unit 1: connecting
ppp_start[1]
ppp phase changed[1]: phase=6
D (5930) esp-netif_lwip-ppp: Phase Establish
pppos_send_config[1]: out_accm=FF FF FF FF
ppp_send_config[1]
pppos_recv_config[1]: in_accm=FF FF FF FF
ppp_recv_config[1]
ppp: auth protocols: PAP=0
sent [LCP ConfReq id=0x1 <asyncmap 0x0> <magic 0xd5b8b5af> <pcomp> <accomp>]
pppos_write[1]: len=24
ppp_start[1]: finished
D (5950) esp_netif_lwip: check: local, if=0x3fc9bd8c fn=0x42013b64
0x42013b64: esp_netif_update_default_netif_lwip at /Users/emanuele/esp/v5.3/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:317

D (5960) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3fc9bd8c
D (5970) esp_netif_lwip: call api in lwip: ret=0x0, give sem
I (5970) test: Waiting for IP address
sent [LCP ConfReq id=0x1 <asyncmap 0x0> <magic 0xd5b8b5af> <pcomp> <accomp>]
pppos_write[1]: len=24
pppos_input[1]: got 21 bytes
pppos_input[1]: Dropping ACCM char <13>
pppos_input[1]: Dropping ACCM char <10>
pppos_input[1]: Dropping ACCM char <13>
pppos_input[1]: Dropping ACCM char <10>
sent [LCP ConfReq id=0x1 <asyncmap 0x0> <magic 0xd5b8b5af> <pcomp> <accomp>]
pppos_write[1]: len=24
sent [LCP ConfReq id=0x1 <asyncmap 0x0> <magic 0xd5b8b5af> <pcomp> <accomp>]
pppos_write[1]: len=24
sent [LCP ConfReq id=0x1 <asyncmap 0x0> <magic 0xd5b8b5af> <pcomp> <accomp>]
pppos_write[1]: len=24
sent [LCP ConfReq id=0x1 <asyncmap 0x0> <magic 0xd5b8b5af> <pcomp> <accomp>]
pppos_write[1]: len=24
sent [LCP ConfReq id=0x1 <asyncmap 0x0> <magic 0xd5b8b5af> <pcomp> <accomp>]
pppos_write[1]: len=24
sent [LCP ConfReq id=0x1 <asyncmap 0x0> <magic 0xd5b8b5af> <pcomp> <accomp>]
pppos_write[1]: len=24
sent [LCP ConfReq id=0x1 <asyncmap 0x0> <magic 0xd5b8b5af> <pcomp> <accomp>]
pppos_write[1]: len=24
sent [LCP ConfReq id=0x1 <asyncmap 0x0> <magic 0xd5b8b5af> <pcomp> <accomp>]
pppos_write[1]: len=24
LCP: timeout sending Config-Requests
ppp phase changed[1]: phase=12
D (65950) esp-netif_lwip-ppp: Phase Disconnect
Connection terminated.
ppp_link_terminated[1]
ppp_link_end[1]
ppp phase changed[1]: phase=0
D (65950) esp-netif_lwip-ppp: Phase Dead
I (65960) esp-netif_lwip-ppp: Connection lost
D (65960) event: running post IP_EVENT:7 with handler 0x4200ac74 and context 0x3fc9bd38 on loop 0x3fc9adec
0x4200ac74: on_ip_event at /Users/emanuele/Documents/Projects/nbTest/main/main.c:29

ppp_link_terminated[1]: finished.
I (65970) test: IP event! 7
I (65980) test: Modem Disconnect from PPP Server
D (65980) event: running post IP_EVENT:7 with handler 0x420128ac and context 0x3fca0818 on loop 0x3fc9adec
0x420128ac: esp_netif_action_disconnected at /Users/emanuele/esp/v5.3/esp-idf/components/esp_netif/esp_netif_handlers.c:83

D (65990) esp_netif_handlers: esp_netif action disconnected with netif0x3fc9bd8c from event_id=7
D (66000) esp_netif_lwip: check: remote, if=0x3fc9bd8c fn=0x42013148
0x42013148: esp_netif_down_api at /Users/emanuele/esp/v5.3/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1673

D (66010) esp_netif_lwip: esp_netif_down_api esp_netif:0x3fc9bd8c
D (66010) esp_netif_lwip: esp_netif_internal_dhcpc_cb lwip-netif:0x3fc9be14
D (66020) esp_netif_lwip: check: local, if=0x3fc9bd8c fn=0x42013b64
0x42013b64: esp_netif_update_default_netif_lwip at /Users/emanuele/esp/v5.3/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:317

D (66030) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3fc9bd8c
D (66030) esp_netif_lwip: call api in lwip: ret=0x0, give sem
```

david-cermak, 2024-08-20:
The PPP negotiation doesn't work, since the modem is still in the text mode:
```
pppos_input[1]: Dropping ACCM char <13>
pppos_input[1]: Dropping ACCM char <10>
```

Also, it rejected the dial-up command for some reason:
```
D (5870) command_lib: generic_command command ATD*99#

D (5870) command_lib: Response: 
ERROR
```

I haven't noticed these lines in the previous log...

ezamp, 2024-08-20:
The SIM7028 module does not implement the ATD*99# command.

david-cermak, 2024-08-23:
> The SIM7028 module does not implement the ATD*99# command.

Oh, sorry, this device does not really seem to support PPP connection. Not possible to acquire an IP after PPPoS handshake then.
You'd need to use AT commands only to connect to internet, it seems.
Looking at the socket TCP/IP commands, they look similar to those of SIMCOM/SIM7600, so you can check this example:
https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/examples/modem_tcp_client
it uses AT commands (might need to create/update them per your device specs) to bring the esp_netif connection and enables using IDF protocol libraries (which might be useful in some cases)

closed 2024-09-28 by ezamp