espressif/esp-protocols #639: Getting "ESP_ERR_TIMEOUT" on `esp_modem_get_signal_quality` call when running example code

gamename, 2024-08-22:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### What component are you using? If you choose Other, provide details in More Information.

esp_modem

### component version

1.1.0

### IDF version.

5.2.1

### More Information.

Hi,

First, let me say this is my first time working with cellular modems, so its probable there is a misconfiguration somewhere...

I'm using a SIM7600X HAT with an ESP32-S3-DevKitC-1-N16. When running the example code (see below), I get this error:
```plaintext
E (00:00:14.767) pppos_example: esp_modem_get_signal_quality failed with 263 ESP_ERR_TIMEOUT
```

The SIM7600X is meant as a HAT for Raspberry Pi (RPi), but can be used by other microcontrollers (like the ESP32).  I confirmed this with the vendor. 

Here are the relevant debugging datapoints:

1. I am using [this](https://github.com/espressif/esp-protocols/blob/master/components/esp_modem/examples/pppos_client/main/pppos_client_main.c) example code. 
2. There is a UART connection between my ESP32 and the SIM7600X.
3. The SIM7600X has 2 interfaces, one for RPi and one for Non-RPi connections. I'm using the non-RPi connections (aka the *yellow* pins).
4. The SIM7600X on-board jumpers are set to the "B" setting.
5. I am using GPIO 17 (Tx) and GPIO 16 (Rx) on the ESP32
6. I also connect GND from the ESP32 to GND on the SIM7600X
7. My provider is Hologram. While running the example app, Hologram shows the board as connected, so the SIM7600X board itself is working. 
8. The SIM7600X board LEDs are illuminated.

Here is output from the ESP32 log:
```
idf monitor
Executing action: monitor
Serial port /dev/cu.usbmodem21101
Connecting...
Detecting chip type... ESP32-S3
Running idf_monitor in directory /Users/tennis/src/C2DS-esp32-coop-controller
Executing "/Users/tennis/.espressif/python_env/idf5.2_py3.12_env/bin/python /Users/tennis/esp/esp-idf/tools/idf_monitor.py -p /dev/cu.usbmodem21101 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 --revision 0 /Users/tennis/src/C2DS-esp32-coop-controller/build/firmware.elf -m '/Users/tennis/.espressif/python_env/idf5.2_py3.12_env/bin/python' '/Users/tennis/esp/esp-idf/tools/idf.py'"...
--- esp-idf-monitor 1.4.0 on /dev/cu.usbmodem21101 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x29 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048d00
0x40048d00: uart_rx_one_char in ROM

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x2da0
SHA-256 comparison failed:
Calculated: 92b6991ae9c9ceb6d9c17ffaa1bba262a31f484d5c3b10fb0c04d18c8a14b89b
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c9914
[... 110 lines cut ...]
D (594) intr_alloc: Connected src 58 to int 3 (cpu 1)
I (00:00:14.152) main_task: Started on CPU0
D (604) heap_init: New heap initialised at 0x3fce9710
D (604) intr_alloc: Connected src 52 to int 13 (cpu 0)
I (00:00:14.182) main_task: Calling app_main()
D (00:00:14.188) esp_netif_lwip: LwIP stack has been initialized
D (00:00:14.194) esp_netif_lwip: esp-netif has been successfully initialized
D (00:00:14.201) event: running task for loop 0x3fc9e898
D (00:00:14.206) event: created task for loop 0x3fc9e898
D (00:00:14.211) event: created event loop 0x3fc9e898
D (00:00:14.217) esp_netif_lwip: check: remote, if=0x3fc9bc60 fn=0x4201c5f8
0x4201c5f8: esp_netif_new_api at /Users/tennis/esp/esp-idf-v5.2.1/components/esp_netif/lwip/esp_netif_lwip.c:690

D (00:00:14.224) esp_netif_objects: esp_netif_add_to_list_unsafe netif added successfully (total netifs: 1)
ppp phase changed[1]: phase=0
D (00:00:14.236) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3fc9fa68
D (00:00:14.244) esp-netif_lwip-ppp: Phase Dead
D (00:00:14.249) esp_netif_lwip: call api in lwip: ret=0x0, give sem
I (00:00:14.256) pppos_example: Initializing esp_modem for the SIM7600 module...
I (00:00:14.264) uart: queue free spaces: 30
D (704) intr_alloc: Connected src 28 to int 17 (cpu 0)
E (00:00:14.767) pppos_example: esp_modem_get_signal_quality failed with 263 ESP_ERR_TIMEOUT
I (00:00:14.768) main_task: Returned from app_main()
```
[image]

david-cermak, 2024-08-23:
Hi @gamename 

I'd suggest that you start with this example: https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/examples/modem_console

This helps fixing the initial connection issues, like wires, baudrate, power supply etc. Since the problem you're experiencing suggests that the modem and the ESP32 cannot exchange a basic AT request/reply.

gamename, 2024-08-23:
> Hi @gamename
>
> I'd suggest that you start with this example: https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/examples/modem_console
>
> This helps fixing the initial [...]

Will do.  Thanks.

gamename, 2024-08-30:
Abandoned use of the Waveshare equipment. Switched to Sixfab

closed 2024-08-30 by gamename