espressif/esp-protocols #646: [mDNS] Query doesn't work

Moddingear, 2024-09-08:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to get the IP of my linux server though mDNS.
I tried using both esp-idf and arduino libraries, and both don't work : The mDNS packet is sent correctly by the esp32, the response is sent by the server, but the esp32 doesn't seem to capture it (checked with Wireshark).
I'm using platformio and an esp32c3, with espressif32 version 6.5.0, framework-arduinoespressif32 version 3.20014.231204 (2.0.14).

Here is a simple code to reproduce the issue, you'll need to change the WiFi SSID and passphrase, as well as the target name.
```cpp
#include <Arduino.h>
#include <WiFi.h>
#include <ESPmDNS.h>
#include <esp_log.h>

#define MODNAME "HealthCheck"

void setup() {
	delay(10*1000);
	ESP_LOGI(MODNAME, "Start health check !");
	WiFi.mode(WIFI_STA);
	WiFi.begin("WIFIName", "WifiPassword");
	ESP_LOGI(MODNAME, "Wifi OK !");
	MDNS.begin("HealthCheck");
}

void loop() {
	if (WiFi.status() != WL_CONNECTED)
	{
		ESP_LOGI(MODNAME, "Wifi not connected !");
		delay(1000);
		return;
	}

	struct esp_ip4_addr addr = {0};

	const char* target_mdns = "TARGET_NAME";

	ESP_LOGI(MODNAME, "Sending request for %s", target_mdns);
	auto target_addr = MDNS.queryHost(target_mdns);
	ESP_LOGI(MODNAME, "Got IP for %s = %s", target_mdns, target_addr.toString());

	delay(10*1000);
}```

david-cermak, 2024-09-09:
Hi @Moddingear 

Since this is probably the most common usecase, I'd like to ask you to check the example here, using the mdns library directly:

https://github.com/espressif/esp-protocols/blob/73c48307a367565f69e724df8641f8cbac45251d/components/mdns/examples/query_advertise

and simply update the host name you're querying here:

https://github.com/espressif/esp-protocols/blob/73c48307a367565f69e724df8641f8cbac45251d/components/mdns/examples/query_advertise/main/mdns_example_main.c#L276

Could you also enable `MDNS_ENABLE_DEBUG_PRINTS=y` and share the log here?

Moddingear, 2024-09-09:
Hello, I tried to compile the test with platformio, but protocol_examples_common.h is nowhere to be found in the repository, and that example is far more complicated than the reproducible code I gave you.

david-cermak, 2024-09-09:
The reason I asked for a reproducer with IDF is that it's possible to update component Kconfig

> Could you also enable MDNS_ENABLE_DEBUG_PRINTS=y and share the log here?

As said above, the code you posted only queries an external host, which is the very basic use-case tested automatically on every commit, so it's unlikely that something would be broken in the library. That's why I would like to rule out some issues with platformio or arduino.

>  but protocol_examples_common.h is nowhere to be found in the repository

This is a small helper component in IDF: https://github.com/espressif/esp-idf/tree/59e18382702b6986be3d3f55e9ac7763c1397cf7/examples/common_components/protocol_examples_common

david-cermak, 2024-10-31:
Closing as not reproducible.

Have installed Arduino and updated your loop function to resolve my MacBook's address:
```cpp
void loop() {

	struct esp_ip4_addr addr = {0};

	const char* target_mdns = "Martins-MacBook-Pro-2";

	ESP_LOGE(MODNAME, "Sending request for %s", target_mdns);
	auto target_addr = MDNS.queryHost(target_mdns);
	ESP_LOGE(MODNAME, "Got IP for %s = %s", target_mdns, target_addr.toString());

	delay(10*1000);
}
```

working as expected:
```
[244753][E][sketch_oct31a.ino:28] loop(): [HealthCheck] Sending request for Martins-MacBook-Pro-2
[245087][E][sketch_oct31a.ino:30] loop(): [HealthCheck] Got IP for Martins-MacBook-Pro-2 = 192.168.0.37
```

closed 2024-10-31 by david-cermak