espressif/esp-protocols #659: Example Request for SIMCOM SIM7070 Switching between GNSS and mobile network connection

timom25ns, 2024-09-18:
### Is your feature request related to a problem?

I'm using ESP-modem to connect to SIMCOM SIM7070G. While on paper it looks as a well-rounded module, a few issues prevent me from using it the way I would like to. 

- an otherwise undocumented issue where GNSS and mobile network connection cannot work concurrently, supposedly due to shared radio parts https://stackoverflow.com/questions/61857667/sim7080g-module-cant-send-data-over-tcp-while-using-gnss 

-it's impossible to enter CMUX mode repeatedly for SIMCOM SIM7070G #168 https://github.com/espressif/esp-protocols/issues/168#issuecomment-1286742413 , and will result in NO-CARRIER responses. 

Some attention was already given to SIM7070, since it contains the specific GNSS command and also an example for processing), but there seems to be very little written about such a concurrent use with a mobile network. 

### Describe the solution you'd like.

If there was such an example within esp-modem, which is dealing with this issue, it would be really amazing. It would also be really helpful if there was a way of getting the GNSS data without leaving CMUX mode, maybe by doing some manual calls which would be otherwise taken care of by the esp_modem mode selector.

timom25ns, 2024-09-18:
@diplfranzhoepfinger

david-cermak, 2024-09-19:
@timom25ns I'll take a look at it, already got one SIM7070G for testing.

> -it's impossible to enter CMUX mode repeatedly for SIMCOM SIM7070G https://github.com/espressif/esp-protocols/issues/168 https://github.com/espressif/esp-protocols/issues/168#issuecomment-1286742413 , and will result in NO-CARRIER responses.

I think this could be fixed or (worked around) easily using manual transitions.

> GNSS and mobile network connection cannot work concurrently,

Not sure how to handle this, but might work if we **pause** the network on our side between GNSS commands?