espressif/esp-protocols #665: Add esp_websocket_client from ESP Register to Project (VSCode)

Oleg-Perevyshin, 2024-09-30:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Traceback (most recent call last):
  File "C:\Users\perevyshin_oe\esp\v5.3.1\esp-idf\tools\idf.py", line 844, in <module>

Executing action: add-dependency

    main()
  File "C:\Users\perevyshin_oe\esp\v5.3.1\esp-idf\tools\idf.py", line 734, in main
    cli(argv, prog_name=PROG, complete_var=SHELL_COMPLETE_VAR)
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\click\core.py", line 1157, in __call__

    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\click\core.py", line 1078, in main

    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\click\core.py", line 1720, in invoke

    return _process_result(rv)
           ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\click\core.py", line 1657, in _process_result
    value = ctx.invoke(self._result_callback, value, **ctx.params)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\click\core.py", line 783, in invoke

    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\esp\v5.3.1\esp-idf\tools\idf.py", line 629, in execute_tasks

    task(ctx, global_args, task.action_args)
  File "C:\Users\perevyshin_oe\esp\v5.3.1\esp-idf\tools\idf.py", line 201, in __call__

    self.callback(self.name, context, global_args, **action_args)
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_manager\idf_extensions.py", line 92, in callback
    getattr(manager, str(subcommand_name).replace('-', '_'))(**kwargs)
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 98, in wrapper

    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 283, in add_dependency
    manifest_filepath, _ = self._get_manifest(component=component, path=path)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 211, in _get_manifest

    manifest_dir = self._get_manifest_dir(component=component, path=path)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 176, in _get_manifest_dir
    elif is_component(Path.cwd()):
         ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_tools\build_system_tools.py", line 111, in is_component

    for line in f:
  File "encodings\cp1251.py", line 23, in decode

UnicodeDecodeError: 'charmap' codec can't decode byte 0x98 in position 231: character maps to <undefined>

Traceback (most recent call last):
  File "C:\Users\perevyshin_oe\esp\v5.3.1\esp-idf\tools\idf.py", line 844, in <module>

Executing action: add-dependency

    main()
  File "C:\Users\perevyshin_oe\esp\v5.3.1\esp-idf\tools\idf.py", line 734, in main
    cli(argv, prog_name=PROG, complete_var=SHELL_COMPLETE_VAR)
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\click\core.py", line 1157, in __call__

    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\click\core.py", line 1078, in main

    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\click\core.py", line 1720, in invoke

    return _process_result(rv)
           ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\click\core.py", line 1657, in _process_result

    value = ctx.invoke(self._result_callback, value, **ctx.params)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\click\core.py", line 783, in invoke

    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\esp\v5.3.1\esp-idf\tools\idf.py", line 629, in execute_tasks

    task(ctx, global_args, task.action_args)
  File "C:\Users\perevyshin_oe\esp\v5.3.1\esp-idf\tools\idf.py", line 201, in __call__

    self.callback(self.name, context, global_args, **action_args)
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_manager\idf_extensions.py", line 92, in callback
    getattr(manager, str(subcommand_name).replace('-', '_'))(**kwargs)
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 98, in wrapper

    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 283, in add_dependency

    manifest_filepath, _ = self._get_manifest(component=component, path=path)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 211, in _get_manifest
    manifest_dir = self._get_manifest_dir(component=component, path=path)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 176, in _get_manifest_dir

    elif is_component(Path.cwd()):
         ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\perevyshin_oe\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_tools\build_system_tools.py", line 111, in is_component

    for line in f:
  File "encodings\cp1251.py", line 23, in decode

UnicodeDecodeError: 'charmap' codec can't decode byte 0x98 in position 231: character maps to <undefined>

Oleg-Perevyshin, 2025-07-08:
Hi!
The problem was in the server settings
I created a project from my company's network
The administrators said that the problem was either with certificates or with DNS servers (I don't remember)
They did something and everything worked well

glmfe, 2025-07-08:
Glad to hear, thanks for the feedback!