espressif/esp-protocols #666: [websocket] Panic abort triggered when sending message chunk

senad-zaimovic, 2024-10-01:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When sending WebSocket messages in chunks repeatedly over time, I eventually get the following error. I am currently using websocket_client library version 1.2.2, but I get the same behaviour with version 1.2.3. Version of ESP-IDF is 5.1.4, but the system is behaving the same even if updated to version 5.3.1.

The logic for sending message in chunks is as following:
`static bool send_websocket_message_in_chunks(const char* message, int32_t message_length)
{
    int32_t message_sent_length = 0;
    bool chunk_sent_successfully = false;

    ESP_LOGD(CONFIG_APP_NETWORK_TASK_NAME, "Sending websocket message in chunks. Message length %ld bytes.",
             message_length);

    /* Send the start of the message */
    if (s_websocket_client_connected) {
        chunk_sent_successfully = 0 == esp_websocket_client_send_text_partial(s_websocket_client, NULL, 0,
                                  WEBSOCKET_MESSAGE_TIMEOUT_TICKS);
        app_watchdog_yield();
    }

    /* Send the remaining message body in chunks */
    while (chunk_sent_successfully && (message_length - message_sent_length > 0)) {
        if (s_websocket_client_connected) {
            int32_t chunk_length = MIN((message_length - message_sent_length), WEBSOCKET_BUFFER_SIZE);

            chunk_sent_successfully = chunk_length == esp_websocket_client_send_cont_msg(
                                          s_websocket_client,
                                          &message[message_sent_length],
                                          chunk_length,
                                          WEBSOCKET_MESSAGE_TIMEOUT_TICKS
                                      );

            if (chunk_sent_successfully) {
                message_sent_length += chunk_length;
            }

            app_watchdog_yield();
        } else {
            chunk_sent_successfully = false;
        }
    }

    if (chunk_sent_successfully) {
        /* Finalize message */
        app_watchdog_yield();
        if (s_websocket_client_connected) {
            chunk_sent_successfully = 0 == esp_websocket_client_send_fin(s_websocket_client, WEBSOCKET_MESSAGE_TIMEOUT_TICKS);
        } else {
            chunk_sent_successfully = false;
        }
    }

    if (!chunk_sent_successfully) {
        if (!s_websocket_client_connected) {
            ESP_LOGE(CONFIG_APP_NETWORK_TASK_NAME, "Websocket client got disconnected during sending message in chunks");
        }

        ESP_LOGE(CONFIG_APP_NETWORK_TASK_NAME, "Failed to send websocket message in chunks. Sent length %ld/%ld bytes.",
                 message_sent_length, message_length);
    } else {
        ESP_LOGD(CONFIG_APP_NETWORK_TASK_NAME, "Completed sending websocket message in chunks. Sent length %ld/%ld bytes.",
                 message_sent_length, message_length);
    }

    return chunk_sent_successfully;
}`

`0x4037a4fe: panic_abort at /opt/esp/idf/components/esp_system/panic.c:463
0x4038d66d: esp_system_abort at /opt/esp/idf/components/esp_system/port/esp_system_chip.c:92
0x40396e01: __assert_func at /opt/esp/idf/components/newlib/assert.c:80
0x40394c89: tlsf_free at /opt/esp/idf/components/heap/tlsf/tlsf.c:1201 (discriminator 1)
0x40394b47: multi_heap_free_impl at /opt/esp/idf/components/heap/multi_heap.c:235
 (inlined by) multi_heap_free_impl at /opt/esp/idf/components/heap/multi_heap.c:224
0x4037aedd: heap_caps_free at /opt/esp/idf/components/heap/heap_caps_base.c:70
0x40396e59: free at /opt/esp/idf/components/newlib/heap.c:39
0x4209e0a1: esp_tls_internal_event_tracker_destroy at /opt/esp/idf/components/esp-tls/esp_tls_error_capture.c:47
0x4209de2b: esp_tls_conn_destroy at /opt/esp/idf/components/esp-tls/esp_tls.c:160
0x420a0b1b: base_close at /opt/esp/idf/components/tcp_transport/transport_ssl.c:320
0x4210acc9: esp_transport_close at /opt/esp/idf/components/tcp_transport/transport.c:172
0x420a11e0: ws_close at /opt/esp/idf/components/tcp_transport/transport_ws.c:645
0x4210acc9: esp_transport_close at /opt/esp/idf/components/tcp_transport/transport.c:172
0x4204ef89: esp_websocket_client_abort_connection at /workspace/packages/hbd/managed_components/espressif__esp_websocket_client/esp_websocket_client.c:226
0x4204f155: esp_websocket_client_send_with_exact_opcode at /workspace/packages/hbd/managed_components/espressif__esp_websocket_client/esp_websocket_client.c:578
0x42050201: esp_websocket_client_send_cont_msg `

bryghtlabs-richard, 2024-12-17:
Seems like heap corruption of something nearby (possibly caused earlier) being detected later when closing the WebSocket during a send(perhaps closing due to heap corruption).

glmfe, 2025-08-25:
Hello @senad-zaimovic,

Thanks for submitting this ticket. I’ll be handling it from now on. Could you confirm if you’re still experiencing the issue?

If so, please let me know which is the latest idf version you're using and the websocket as well.

senad-zaimovic, 2025-09-02:
@glmfe It seems to be a bit better with the latest version of the Websocket client. I will keep an eye on it, but so far I haven't experienced an issue with a new version.