espressif/esp-protocols #674: esp_modem at_raw has the wrong length output string

txf-, 2024-10-18:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm executing the following function:

```
  void getGps(){
    std::string out;
    command_result res = dce->at_raw("AT+CGNSINF\r",out,"+CGNSINF:","ERROR", 10000);
    ESP_LOGI(COMM_TAG,"get gps str: %s", out.c_str()); 
  }
```

And the response output I get is this (gps coords hidden) :
`+CGNSINF: 1,1,20241018103437.000,xx.yyyyyy,-x.yyyyyy,-11.900,0`
or this:
`+CGNSINF: 1,1,20241018103421.000,xx.yyyyyy,-x.yyyyyy,95.000,0.`

its seems the string length is of a fixed size, the whole string should be something like this:
`+CGNSINF: 1,1,20241018103758.000,xx.yyyyyyy,-x.yyyyyy,15.300,0.00,4.8,1,,1.1,1.5,0.9,,27,4,3,,37,,`

is there a way to set the response buffer size?

david-cermak, 2024-10-25:
@txf- Are your really calling the C++ API and not the `esp_modem_at_raw()` plain C wrapper? The C functions indeed have fixed output length parameter, but C++ use standard strings which could grow.

Isn't is possible that the response comes in chunks? Your *pass criteria* (`"+CGNSINF:"`) are located in the beginning of the reply so that when the first chunk comes, all's good for returning `OK` with the partial response.

Could you please try:
```cpp
  void getGps(){
    std::string out;
    command_result res = dce->at_raw("AT+CGNSINF\r",out,"SomethingThatNeverComes:","ERROR", 10000);
    ESP_LOGI(COMM_TAG,"get gps str: %s", out.c_str()); 
  }
```
Now this should fail, but once it would, should include the entire string.

txf-, 2024-11-05:
@david-cermak 
I changed it to look for the "OK", which is sent a few moments after the string. It then captures the whole string.

closed 2024-11-05 by txf-