espressif/esp-protocols #682: [websocket] keepalives do not work over SSL over TCP

bryghtlabs-richard, 2024-10-30:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Keepalive does not work when websocket client is used over SSL(or TLS) over TCP.

I would expect the following flags added to my websocke would reach past the SSL/TLS layer down to the TCP layer and configure TCP keepalives, but they do not. Instead they seem to do nothing.

    wsCfg.keep_alive_enable = true;       /*!< Enable keep-alive timeout */
    wsCfg.keep_alive_idle = 1;            /*!< Keep-alive idle time. Default is 5 (second) */
    wsCfg.keep_alive_interval = 2;        /*!< Keep-alive interval time. Default is 5 (second) */
    wsCfg.keep_alive_count = 20;          /*!< Keep-alive packet retry send count. Default is 3 counts */

bryghtlabs-richard, 2024-10-30:
Looks like this was fixed in c728eae5ea9fdebb4f2ea821aa8db32c3b16e755

bryghtlabs-richard, 2024-10-30:
Please consider this as a request to update https://components.espressif.com/components/espressif/esp_websocket_client

david-cermak, 2024-10-31:
Thanks for reminding! The component version hasn't been updated for a long time, I'll publish it this week, after merging some minor fixes that've been approved already.

david-cermak, 2024-10-31:
websocket client `v1.3.0` [published](https://components.espressif.com/components/espressif/esp_websocket_client/versions/1.3.0/changelog?language=en)

closed 2024-10-31 by david-cermak