espressif/esp-protocols #706: esp-modem: esp_modem_set_urc buffer issues? seeing same URCs multiple times

jodab, 2024-12-05:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I decided to test using the new urc callback feature but running in to some strange issues.

The esp_modem_set_urc(dce, urc_handler); is returning the same lines (or chunks of data) several times, from how I understand it. I've tried just printing whatever is returned without any kind of processing and the duplicate messages keeps coming. I'm not the most experienced with c programming so I might have misunderstood something here..

But in essence it's like every read seems to include previous data still in the buffer.

I verified that the modem isn't doing some funny stuff using an a different app on the esp32 which just let me to raw read/writes directly to the uart, and no problems there.

I've attached the code if it is of any help.

Here's a log while trying to debug what's going on:

`I (00:00:09.032) main: Waiting for modem to respond...
I (00:00:09.532) main: Waiting for modem to respond...
Raw UART data (15 bytes): 0D 0A 2A 41 54 52 45 41 44 59 3A 20 31 0D 0A 
I (00:00:09.999) main: URC received at 9916 ms: 
*ATREADY: 1

I (00:00:10.001) main: Current URC Buffer: '
*ATREADY: 1
'
I (00:00:10.007) main: Ignored Non-URC Line: 
I (00:00:10.012) main: Ignored Non-URC Line: *ATREADY: 1
I (00:00:10.018) main: Buffer after processing: ''
I (00:00:10.032) main: Waiting for modem to respond...
Raw UART data (9 bytes): 41 54 0D 0D 0A 4F 4B 0D 0A 
I (00:00:10.041) main: URC received at 9958 ms: AT
OK

I (00:00:10.041) main: Current URC Buffer: 'AT
OK
'
I (00:00:10.041) main: Modem synchronized successfully.
I (00:00:10.046) main: Ignored Non-URC Line: AT
I (00:00:10.057) main: Ignored Non-URC Line: OK
I (00:00:10.063) main: Buffer after processing: ''
Raw UART data (11 bytes): 41 54 45 30 0D 0D 0A 4F 4B 0D 0A 
I (00:00:11.070) main: URC received at 10987 ms: ATE0
OK

I (00:00:11.071) main: Current URC Buffer: 'ATE0
OK
'
I (00:00:11.071) main: ATE0 success
I (00:00:11.077) main: Ignored Non-URC Line: ATE0
I (00:00:11.086) main: Ignored Non-URC Line: OK
I (00:00:11.092) main: Buffer after processing: ''
Raw UART data (16 bytes): 0D 0A 2B 43 50 49 4E 3A 20 52 45 41 44 59 0D 0A 
I (00:00:11.882) main: URC received at 11799 ms: 
+CPIN: READY

I (00:00:11.884) main: Current URC Buffer: '
+CPIN: READY
'
I (00:00:11.891) main: Ignored Non-URC Line: 
I (00:00:11.896) main: Processing URC: +CPIN: READY
I (00:00:11.902) main: GNSS Info: +CPIN: READY
I (00:00:11.907) main: Buffer after processing: ''
Raw UART data (28 bytes): 0D 0A 2B 43 50 49 4E 3A 20 52 45 41 44 59 0D 0A 0D 0A 53 4D 53 20 44 4F 4E 45 0D 0A 
I (00:00:13.323) main: URC received at 13241 ms: 
+CPIN: READY

SMS DONE

I (00:00:13.330) main: Current URC Buffer: '
+CPIN: READY

SMS DONE
'
I (00:00:13.338) main: Ignored Non-URC Line: 
I (00:00:13.343) main: Processing URC: +CPIN: READY
I (00:00:13.348) main: GNSS Info: +CPIN: READY
I (00:00:13.354) main: Ignored Non-URC Line: 
I (00:00:13.359) main: Ignored Non-URC Line: SMS DONE
I (00:00:13.364) main: Buffer after processing: ''
Raw UART data (41 bytes): 0D 0A 2B 43 50 49 4E 3A 20 52 45 41 44 59 0D 0A 0D 0A 53 4D 53 20 44 4F 4E 45 0D 0A 0D 0A 2B 53 4D 53 20 46 55 4C 4C 0D 0A 
I (00:00:13.442) main: URC received at 13360 ms: 
+CPIN: READY

SMS DONE

+SMS FULL

I (00:00:13.452) main: Current URC Buffer: '
+CPIN: READY

SMS DONE

+SMS FULL
'
I (00:00:13.461) main: Ignored Non-URC Line: 
I (00:00:13.466) main: Processing URC: +CPIN: READY
I (00:00:13.472) main: GNSS Info: +CPIN: READY
I (00:00:13.477) main: Ignored Non-URC Line: 
I (00:00:13.482) main: Ignored Non-URC Line: SMS DONE
I (00:00:13.487) main: Ignored Non-URC Line: 
I (00:00:13.492) main: Processing URC: +SMS FULL
I (00:00:13.498) main: GNSS Info: +SMS FULL
I (00:00:13.503) main: Buffer after processing: ''
Raw UART data (65 bytes): 0D 0A 2B 43 50 49 4E 3A 20 52 45 41 44 59 0D 0A 0D 0A 53 4D 53 20 44 4F 4E 45 0D 0A 0D 0A 2B 53 4D 53 20 46 55 4C 4C 0D 0A 0D 0A 2B 43 47 45 56 3A 20 45 50 53 20 50 44 4E 20 41 43 54 20 31 0D 0A 
I (00:00:16.692) main: URC received at 16609 ms: 
+CPIN: READY

SMS DONE

+SMS FULL

+CGEV: EPS PDN ACT 1

I (00:00:16.703) main: Current URC Buffer: '
+CPIN: READY

SMS DONE

+SMS FULL

+CGEV: EPS PDN ACT 1
'
I (00:00:16.715) main: Ignored Non-URC Line: 
I (00:00:16.720) main: Processing URC: +CPIN: READY
I (00:00:16.725) main: GNSS Info: +CPIN: READY
I (00:00:16.730) main: Ignored Non-URC Line: 
I (00:00:16.735) main: Ignored Non-URC Line: SMS DONE
I (00:00:16.741) main: Ignored Non-URC Line: 
I (00:00:16.746) main: Processing URC: +SMS FULL
I (00:00:16.751) main: GNSS Info: +SMS FULL
I (00:00:16.756) main: Ignored Non-URC Line: 
I (00:00:16.761) main: Processing URC: +CGEV: EPS PDN ACT 1
I (00:00:16.768) main: GNSS Info: +CGEV: EPS PDN ACT 1
I (00:00:16.773) main: Buffer after processing: ''
Raw UART data (76 bytes): 0D 0A 2B 43 50 49 4E 3A 20 52 45 41 44 59 0D 0A 0D 0A 53 4D 53 20 44 4F 4E 45 0D 0A 0D 0A 2B 53 4D 53 20 46 55 4C 4C 0D 0A 0D 0A 2B 43 47 45 56 3A 20 45 50 53 20 50 44 4E 20 41 43 54 20 31 0D 0A 0D 0A 50 42 20 44 4F 4E 45 0D 0A 
I (00:00:22.445) main: URC received at 22362 ms: 
+CPIN: READY

SMS DONE

+SMS FULL

+CGEV: EPS PDN ACT 1

PB DONE

I (00:00:22.458) main: Current URC Buffer: '
+CPIN: READY

SMS DONE

+SMS FULL

+CGEV: EPS PDN ACT 1

PB DONE
'
I (00:00:22.470) main: Ignored Non-URC Line: 
I (00:00:22.475) main: Processing URC: +CPIN: READY
I (00:00:22.481) main: GNSS Info: +CPIN: READY
I (00:00:22.486) main: Ignored Non-URC Line: 
I (00:00:22.491) main: Ignored Non-URC Line: SMS DONE
I (00:00:22.497) main: Ignored Non-URC Line: 
I (00:00:22.502) main: Processing URC: +SMS FULL
I (00:00:22.507) main: GNSS Info: +SMS FULL
I (00:00:22.512) main: Ignored Non-URC Line: 
I (00:00:22.517) main: Processing URC: +CGEV: EPS PDN ACT 1
I (00:00:22.523) main: GNSS Info: +CGEV: EPS PDN ACT 1
I (00:00:22.529) main: Ignored Non-URC Line: 
I (00:00:22.534) main: Ignored Non-URC Line: PB DONE
I (00:00:22.539) main: Buffer after processing: ''`
[main.zip](https://github.com/user-attachments/files/18018713/main.zip)

jodab, 2024-12-06:
Problem went away when enabling CMUX strangely

jodab, 2024-12-06:
> @jodab It is still expected that it keeps buffering up (for easier parsing, as the data might come in chunks, but maybe I could add another parameter to indicate the current chunk?)
> 
> > Problem went away when enabling CMUX strangely
> 
> This is expected, too, as we process the data on CMUX frame boundary (here data come in frames -- could be chunked, but the library defragments it to frames)

Thank you so much for clarifying. I think I just have to rethink how I approach this, but the parameter to indicate current chunk sounds like it would make it easier to work with.

AshUK, 2025-03-04:
@david-cermak Also experiencing this, yes it would be useful to know the current chunk or if the URC is complete message or chunked message. Knowing if a command is complete is useful when issuing the same command twice and you need to invalidate the older responses.

I didn't see the issue go away when using CMUX mode