espressif/esp-protocols #715: [websocket]: ability to read connection upgrade headers

bryghtlabs-richard, 2024-12-17:
### Is your feature request related to a problem?

ESP WebSocket-client instances are unable to use cookie-based sticky sessions, because there is no interface to extract the upgrade response headers. 

Here's an example of what we'd like to use: https://docs.aws.amazon.com/prescriptive-guidance/latest/load-balancer-stickiness/alb-cookies-stickiness.html, which tries to set `lb_cookie` to remember which backend server the device connected to, but there's no way for the client to read the value of the `Set-Cookie: lb_cookie=<server-index-value>` header.

### Describe the solution you'd like.

When connecting, I'd like the ability to iterate HTTP response headers during the WEBSOCKET_EVENT_CONNECTED event handler.

Once I have this, I can save out the `lb_cookie` for use in future reconnections.

bryghtlabs-richard, 2025-08-27:
Will we need some sort of feature-flag from the ESP-IDF to know if `header_hook` is available?

glmfe, 2025-08-28:
Hi @bryghtlabs-richard ,

we might use the IDF version tag, I'm just checking whether this will be backported to 5.5.0

```
#if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(6, 0, 0)
//Code that refers to `header_hook`
#endif
```