espressif/esp-protocols #716: PdpContext not declared when building for a component in c++

atroncoso-tucar, 2024-12-17:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi all, 
I'm working on a project written in c++ and I've been developing a component which serves as an adapter which makes use of the esp_modem component to suit our needs. 
Based on the examples, I've successfully used the esp_modem component, to connect to internet, ping a server or connect to an mqtt broker. 
However, when bringing the implementation into the the larger project, I ran into the following build error: 

```bash
In file included from <path>/managed_components/espressif__esp_modem/include/generate/esp_modem_command_declare.inc:16,
                 from <path>/test/managed_components/espressif__esp_modem/include/esp_modem_api.h:10
...
/home/user/Documents/tucar/github/tucar/Locker-Hub/test/managed_components/espressif__esp_modem/include/generate/esp_modem_command_declare.inc:113:78: error: 'PdpContext' has not been declared
  113 | ESP_MODEM_DECLARE_DCE_COMMAND(set_pdp_context, command_result, 1, STRUCT_OUT(PdpContext, p1)) \
      |                                                                              ^~~~~~~~~~
/home/user/Documents/tucar/github/tucar/Locker-Hub/test/managed_components/espressif__esp_modem/include/generate/esp_modem_command_declare_helper.inc:17:38: note: in definition of macro 'STRUCT_OUT'
   17 | #define STRUCT_OUT(struct_name, p1)  struct_name& p1
      |                                      ^~~~~~~~~~~
/home/user/Documents/tucar/github/tucar/Locker-Hub/test/managed_components/espressif__esp_modem/include/generate/esp_modem_command_declare.inc:113:1: note: in expansion of macro 'ESP_MODEM_DECLARE_DCE_COMMAND'
  113 | ESP_MODEM_DECLARE_DCE_COMMAND(set_pdp_context, command_result, 1, STRUCT_OUT(PdpContext, p1)) \
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/Documents/tucar/github/tucar/Locker-Hub/test/managed_components/espressif__esp_modem/include/esp_modem_api.h:21:1: note: in expansion of macro 'DECLARE_ALL_COMMAND_APIS'
   21 | DECLARE_ALL_COMMAND_APIS(declares esp_modem_<API>(esp_modem_t *dce, ...);)
      | ^~~~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.

```

After inspecting the code and attempting to bring the implementation as close as possible to the examples I had used successfully, the only _relevant_ difference I could find was the fact that one was interpreted as C++ while the examples were interpreted as C. 
With this hypothesis, I began inspecting the module code to try and understand where the differences may lie. 
I noticed that in the include file trail, in c++ the esp_modem_types.hpp defines the missing struct. However, it's nested within the esp_modem namespace. 

Additionally, I found a similar struct is also defined in esp_modem_PdpContext_t is also defined in esp_modem_c_api_types.h so I assume the first is used in the context of C++ while the second in the context of C. 

In short, I know where the struct is defined, but don't know how to include it properly when using esp_modem to construct a component built in C++. 

I've tried reading through the documentation and examples presented but none have given me insights on what I'm doing wrong specifically in the context of this project. 
Could it be that I'm not including the header files correctly? Maybe in the wrong order (though I hope that this isn't the problem). 

Any tips on how to solve this would be great. 
Could it possibly be a build bug?
For now... I'm forward declaring the PdpContext struct before including the header files. Yes, it's ugly but at least its allowed me to keep going so far. 

Any help would be great. 
Thanks!

david-cermak, 2024-12-18:
Do you actually include `esp_modem_api.h` from your C++ wrapper?

You're supposed to include 
```cpp
#include "cxx_include/esp_modem_api.hpp"
```
from C++ projects instead.

The C and C++ API are not interchangeable, but they're using the same generic types behind the scenes (and the macro expansion isn't very straight forward, sorry for that -- already addressing it in a PR)

Or, could you please share some portions (basically include paths) of your project if the `esp_modem_api.h` gets included in some other way?

david-cermak, 2025-01-10:
Closing as the comment above likely resolved the issue. Feel free to reopen if needed.

closed 2025-01-10 by david-cermak