espressif/esp-protocols #723: ESP32-S3 fails to reconnect to MQTT SSL broker with A7670G LTE modem

Sanjana000000, 2024-12-31:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP32 S3 is connected to the A7670G lte modem via usb and the pppos_client is being used from esp_modem external component.ESP32 S3 succesfully connects to the Non-SSL broker but fails to reconnect to the SSL Mqtt broker on disconnect.
During testing, we observed the following issue with MQTT reconnection after a disconnect event. The LTE module otherwise appears to maintain its network connectivity.

The device fails to reconnect, and the logs show the following error:

[0;31mE (2898550) esp-tls: couldn't get hostname for :<mqtt_ssl_broker_address>: getaddrinfo() returns 202, addrinfo=0x0[0m

[0;31mE (2898550) esp-tls: Failed to open new connection[0m

[0;31mE (2898560) transport_base: Failed to open a new connection[0m

[0;31mE (2898560) mqtt_client: Error transport connect[0m

[0;32mI (2898570) APP_MQTT: INFO:MQTT EVENT_0[0m

[0;31mE (2898570) APP_MQTT: INFO:MQTT_EVENT_ERROR[0m

[0;31mE (2898580) APP_MQTT: Last error reported from esp-tls: 0x8001[0m

[0;32mI (2898580) APP_MQTT: Last errno string (Success)[0m

[0;32mI (2898590) APP_MQTT: INFO:MQTT EVENT_2[0m

[0;32mI (2898590) APP_MQTT: INFO:Client disconnected[0m

david-cermak, 2025-01-06:
>  The LTE module otherwise appears to maintain its network connectivity.

Are you sure that the module is still connected? The modem library doesn't automatically reconnect when connection is lost.
Are you getting some events from PPP/esp_netif? Did you see a lost IP event?
Could you please check the connectivity (e.g. calling `ping` with some internet service)?

Sanjana000000, 2025-01-07:
> Are you sure that the module is still connected? The modem library doesn't automatically reconnect when connection is lost.
> Are you getting some events from PPP/esp_netif? Did you see a lost IP event?
> Could you please check the connectivity (e.g. calling `ping` with some internet service)?

- If the connection is lost, we manually reinitialize the LTE module to reconnect to the network and receive the "GOT IP" event.
- Yes, we received some events regarding the lost IP : 
       -**esp-netif_lwip-ppp: Connection lost
       -Modem Disconnect from the PPP Server
       -Lost IP**
- Upon reconnecting, we received a "GOT IP" event.

david-cermak, 2025-01-15:
> esp-tls: couldn't get hostname for :<mqtt_ssl_broker_address>: getaddrinfo() returns 202,

Do you use any other interface of ESP32-S3 (e.g. wifi station of AP) at the same time? Would the re-connection work if you supply IP address directly?

txf-, 2025-10-01:
> > esp-tls: couldn't get hostname for :<mqtt_ssl_broker_address>: getaddrinfo() returns 202,
> 
> Do you use any other interface of ESP32-S3 (e.g. wifi station of AP) at the same time? Would the re-connection work if you supply IP address directly?

I was under the impression that for TLS, one requires a hostname, not an IP.

I'm having the same issue with a SIM7000, it connects to the operator, PPP starts correctly and maintains, but then It completely fails to resolve DNS.

Could it be due to weak signal quality at the limits of what is acceptable? If it is too bad the modem drops the PPP connection, but I notice the prevalence of this happening when the signal is -90 dBm or lower.