espressif/esp-protocols #734: mqtt does not url decode username or password

0xFEEDC0DE64, 2025-01-16:
### Is your feature request related to a problem?

yes, our customers have mqtt servers with usernames and passwords containing spaces, @ and other special characters.

Telling them to correctly urlencode the parameters didnt help.

We only expose one single mqtt connection url and not the individual username and password fields.

### Describe the solution you'd like.

The mqtt library should urldecode automatically if the username and passwords are taken from the url

euripedesrocha, 2025-01-16:
@0xFEEDC0DE64 could you add more details on it? Also this is an issue on esp-mqtt right?

0xFEEDC0DE64, 2025-01-16:
Ah right, is esp-mqtt in the esp-idf repository? I will write there agian