espressif/esp-protocols #754: mDNS resolution fails when requesting SRV and TXT records

tl-sl, 2025-02-01:
### Answers checklist.

- [x] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [x] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Recently in Home Assistant there have been issues when resolving mdns addresses of devices running on esp32.  After some debugging it seems that when requesting `SRV` and `TXT` records along with `A` and `AAAA` from `name.local` the devices fail to respond at all. I believe this is the default in Zeroconf. However if requesting only `A` and `AAAA` record then do correctly get a response.

This affects Arduino v3, but also tested on esp-idf v5.4 with latest mDNS component and using the query_advertise example which has same issue.

Here is a test script that reproduces the issues, running this script no answer is ever received.

```
import asyncio
import logging

from zeroconf import (
    DNSOutgoing,
    DNSQuestion,
)
from zeroconf.asyncio import AsyncZeroconf
from zeroconf.const import (
    _CLASS_IN,
    _FLAGS_QR_QUERY,
    _TYPE_A,
    _TYPE_AAAA,
    _TYPE_ANY,
    _TYPE_TXT,
    _TYPE_SRV,
)

logging.basicConfig(level=logging.DEBUG)
logging.getLogger("zeroconf").setLevel(logging.DEBUG)
name = "esp32-mdns.local."
lower_name = name.lower()

async def test_send_a_aaaa_query_lowercase_a_with_srv_txt():
    """Send a query with the A record type with the lowercase name."""
    aiozc = AsyncZeroconf()
    await aiozc.zeroconf.async_wait_for_start()
    outgoing = DNSOutgoing(_FLAGS_QR_QUERY)
    outgoing.add_question(DNSQuestion(lower_name, _TYPE_SRV, _CLASS_IN))
    outgoing.add_question(DNSQuestion(lower_name, _TYPE_TXT, _CLASS_IN))
    outgoing.add_question(DNSQuestion(lower_name, _TYPE_A, _CLASS_IN))
    outgoing.add_question(DNSQuestion(lower_name, _TYPE_AAAA, _CLASS_IN))
    aiozc.zeroconf.async_send(outgoing)
    await asyncio.sleep(3)

asyncio.run(test_send_a_aaaa_query_lowercase_a_with_srv_txt())
```

However if remove the following 2 lines from that script:
```
    outgoing.add_question(DNSQuestion(lower_name, _TYPE_SRV, _CLASS_IN))
    outgoing.add_question(DNSQuestion(lower_name, _TYPE_TXT, _CLASS_IN))
```

It works as expected and a response is received:
```
DEBUG:zeroconf:Received from '192.168.1.203':5353 [socket 6 (('0.0.0.0', 5353))]: <DNSIncoming:id=0, flags=33792, truncated=False, n_q=0, n_ans=4, n_auth=0, n_add=0, questions=[], answers=[record[a,in-unique,esp32-mdns.local.]=120.0/119,192.168.178.203, record[quada,in-unique,esp32-mdns.local.]=120.0/119,fe80::f612:faff:fe59:c86c, record[quada,in-unique,esp32-mdns.local.]=120.0/119,2400:a842:943b:0:f612:faff:fe59:c86c, record[quada,in-unique,esp32-mdns.local.]=120.0/119,fdf1:effc:d430:0:f612:faff:fe59:c86c]> (128 bytes) as [b'\x00\x00\x84\x00\x00\x00\x00\x04\x00\x00\x00\x00\nesp32-mdns\x05local\x00\x00\x01\x80\x01\x00\x00\x00x\x00\x04\xc0\xa8\xb2\xcb\xc0\x0c\x00\x1c\x80\x01\x00\x00\x00x\x00\x10\xfe\x80\x00\x00\x00\x00\x00\x00\xf6\x12\xfa\xff\xfeY\xc8l\xc0\x0c\x00\x1c\x80\x01\x00\x00\x00x\x00\x10$\x00\xa8B\x94;\x00\x00\xf6\x12\xfa\xff\xfeY\xc8l\xc0\x0c\x00\x1c\x80\x01\x00\x00\x00x\x00\x10\xfd\xf1\xef\xfc\xd40\x00\x00\xf6\x12\xfa\xff\xfeY\xc8l']
```

david-cermak, 2025-02-03:
Hi @tl-sl 

Shouldn't you be asking for the actual service/protocol when querying `SRV` and `TXT` records?

I can see that the ESP32 sees the query packet as:
```
RX[0][0]: From: 192.168.0.32:5353, To: 224.0.0.251, Packet[1494370]: 
    Q: esp32-mdns...local. SRV IN
    Q: esp32-mdns...local. TXT IN
    Q: esp32-mdns.local. A IN
    Q: esp32-mdns.local. AAAA IN
```
which are not the services and TXT records it advertizes:

after correcting these queries
```python
outgoing.add_question(DNSQuestion("_http._tcp.local.", _TYPE_SRV, _CLASS_IN))
outgoing.add_question(DNSQuestion("_http._tcp.local.", _TYPE_TXT, _CLASS_IN))
````
the ESP32 correctly responds.

I'll have to consult the specification on these invalid queries, if we still need to reply, at least to the valid items or not. (it looks like the older versions of mdns lib were more permissive, but need to double check...)

tl-sl, 2025-02-03:
I believe these invalid queries were coming from Zeroconf defaults. Shouldnt mDNS ignore the questions it cant answer and respond to the ones it can?

closed 2025-02-07 by david-cermak

david-cermak, 2025-02-10:
@tl-sl Fixing in https://github.com/espressif/esp-protocols/commit/cd07228f81603103e5b56f2d2334743a9db2f36c 

please comment or reopen if it doesn't work.