espressif/esp-protocols #757: eppp_link wifi and spi ethernet (w5500) don't play nice together

bensuffolk, 2025-02-03:
### Answers checklist.

- [x] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [x] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have 2 ESP32-S3 devices connected to each other with eppp_link using SPI mode (Using host 2).

On the slave ESP (running the eppp_link server) I have a w5500 SPI ethernet net connection (Using host 1).

I a running a http server on each ESP, and have set up a port forward from the slave ESP to the HOST ESP allowing me to happily access both web servers from the wired network that is the w5500.

If I replace the w5500 with the WiFi interface on the slave ESP everything works as it should also.

However if I enable both the w5500 and the WiFi adapters the w5500 works as expected but the WiFi connections drops packets and does not let me connect to the the webserver on the slave (or the host), and once I try to establish a connection to the slave webserver I get errors with the packets between the eppp_link with the sockets being closed and not being able to be opened again with bad file number errors.

I am going to try and pair down to a very simple application with nothing else running other that the 3 networks to make sure I can rule everything else out like memory issues etc.

But if in the mean time anybody has any insight into this I'd be interested.

david-cermak, 2025-02-04:
So the w5500 is only on one ESP32-S3, correct?

Could you please try if SPI master worked better on that device with w5500 ? you can interchangeably run either eppp server or client on both SPI master/slave.
I cannot tell what's wrong now, just assume that SPI master would likely coexist with another SPI master.

bensuffolk, 2025-02-04:
I had been using SPI master fine before switching to eppp_link so I know its not an SPI issue.

I have done some further testing and I'm pretty sure it's down to memory because if I don't run anything else apart from the networking I can happily have eppp_link and WiFi and W5500 running. 

Just a bit surprising that there are no error messages posted by the WiFi driver to tell me whats going on.

I will probably just end up offloading some of the other code onto the second ESP which has a good amount of free PSRAM on it, where as the slave has no PSRAM