espressif/esp-protocols #759: SLIP

nart-schinackow-munoz, 2025-02-10:
### Answers checklist.

- [x] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [x] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I am trying the slip_custom_netif and my esp32s3 keeps rebooting.
My Idea is to use the esp32 connected to a Linux pc via Serial using the "slattach -v -L -s 115200 -p slip /dev/ttyUSB1"

below is the log:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1b34
load:0x403c8700,len:0x4
load:0x403c8704,len:0xfbc
load:0x403cb700,len:0x3820
entry 0x403c89e4
I (27) boot: ESP-IDF v5.3.2-dirty 2nd stage bootloader
I (27) boot: compile time Feb 10 2025 22:04:01
I (27) boot: Multicore bootloader
I (31) boot: chip revision: v0.2
I (34) boot: efuse block revision: v1.3
I (39) qio_mode: Enabling QIO for flash chip WinBond
I (44) boot.esp32s3: Boot SPI Speed : 80MHz
I (49) boot.esp32s3: SPI Mode       : QIO
I (54) boot.esp32s3: SPI Flash Size : 16MB
I (59) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  2 factory          factory app      00 00 00010000 00100000
I (98) boot: End of partition table
I (102) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=10968h ( 67944) map
I (121) esp_image: segment 1: paddr=00020990 vaddr=3fc94000 size=02ddch ( 11740) load
I (124) esp_image: segment 2: paddr=00023774 vaddr=40374000 size=0c8a4h ( 51364) load
I (138) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=323e0h (205792) map
I (172) esp_image: segment 4: paddr=00062408 vaddr=403808a4 size=036c4h ( 14020) load
I (188) boot: Loaded app from partition at offset 0x10000
I (188) boot: Disabling RNG early entropy source...
I (200) octal_psram: vendor id    : 0x0d (AP)
I (200) octal_psram: dev id       : 0x02 (generation 3)
I (200) octal_psram: density      : 0x03 (64 Mbit)
I (205) octal_psram: good-die     : 0x01 (Pass)
I (210) octal_psram: Latency      : 0x01 (Fixed)
I (216) octal_psram: VCC          : 0x01 (3V)
I (221) octal_psram: SRF          : 0x01 (Fast Refresh)
I (226) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (232) octal_psram: BurstLen     : 0x01 (32 Byte)
I (238) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (244) octal_psram: DriveStrength: 0x00 (1/1)
I (249) esp_psram: Found 8MB PSRAM device
I (254) esp_psram: Speed: 40MHz
I (258) cpu_start: Multicore app
I (992) esp_psram: SPI SRAM memory test OK
I (1001) cpu_start: Pro cpu start user code
I (1001) cpu_start: cpu freq: 240000000 Hz
I (1001) app_init: Application information:
I (1004) app_init: Project name:     slip_test
I (1010) app_init: App version:      1
I (1014) app_init: Compile time:     Feb 10 2025 22:13:24
I (1020) app_init: ELF file SHA256:  c155eb792...
I (1025) app_init: ESP-IDF:          v5.3.2-dirty
I (1031) efuse_init: Min chip rev:     v0.0
I (1036) efuse_init: Max chip rev:     v0.99 
I (1041) efuse_init: Chip rev:         v0.2
I (1046) heap_init: Initializing. RAM available for dynamic allocation:
I (1053) heap_init: At 3FC980C0 len 00051650 (325 KiB): RAM
I (1059) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (1065) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1072) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (1078) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (1086) spi_flash: detected chip: winbond
I (1090) spi_flash: flash io: qio
I (1094) sleep: Configure to isolate all GPIO pins in sleep state
I (1101) sleep: Enable automatic switching of GPIO sleep configuration
I (1109) main_task: Started on CPU0
I (1129) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1129) main_task: Calling app_main()
I (1129) slip-example: Initialising SLIP interface
I (1139) slip-example: Initialising SLIP modem
I (1139) slip-modem: slip_modem_create: Creating slip modem (netif: 0x3fca6b00)
I (1149) uart: queue free spaces: 10
I (1159) slip-modem-netif: slip_modem_netif_start: Starting SLIP interface: lwip netif:0x3fca6b88
I (1159) slip-example: SLIP init complete
I (1169) slip-example: Socket bound, port 5678
I (1169) slip-example: Starting node manager UDP task
I (1179) main_task: Returned from app_main()
I (1259) slip-modem: esp_netif 0x3fca6b00
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x420131c1  PS      : 0x00060a30  A0      : 0x8202092c  A1      : 0x3fca7f30  
--- 0x420131c1: pbuf_cat at /esp/v5.3.2/esp-idf/components/lwip/lwip/src/core/pbuf.c:861 (discriminator 1)

A2      : 0x0200000a  A3      : 0x3fc99b24  A4      : 0x3fc99b24  A5      : 0x00000000  
A6      : 0x3fc9a124  A7      : 0x00000000  A8      : 0x000005de  A9      : 0x0200000a  
A10     : 0x000005de  A11     : 0x3fc9a134  A12     : 0x00000002  A13     : 0x00000002  
A14     : 0x00000182  A15     : 0x00000000  SAR     : 0x00000006  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0200000a  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffd  
--- 0x400556d5: strlen in ROM
0x400556e5: strlen in ROM

Backtrace: 0x420131be:0x3fca7f30 0x42020929:0x3fca7f50 0x4202099c:0x3fca7f70 0x42020aef:0x3fca7f90 0x4200c8da:0x3fca7fb0 0x420300d1:0x3fca7fd0 0x4200c6a1:0x3fca7ff0 0x4037beed:0x3fca8010
--- 0x420131be: pbuf_cat at /esp/v5.3.2/esp-idf/components/lwip/lwip/src/core/pbuf.c:863
0x42020929: slipif_rxbyte at /esp/v5.3.2/esp-idf/components/lwip/lwip/src/netif/slipif.c:276
0x4202099c: slipif_rxbyte_enqueue at/esp/v5.3.2/esp-idf/components/lwip/lwip/src/netif/slipif.c:494
0x42020aef: slipif_received_bytes /esp/v5.3.2/esp-idf/components/lwip/lwip/src/netif/slipif.c:555
0x4200c8da: esp_netif_lwip_slip_input at /dev/slip_test/main/slip_modem_netif.c:71
0x420300d1: esp_netif_receive at /esp/v5.3.2/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1278
0x4200c6a1: slip_modem_uart_rx_task at /dev/slip_test/main/slip_modem.c:238
0x4037beed: vPortTaskWrapper at /esp/v5.3.2/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

Am I missing something here ? or Am I supposed to to do any extra configuration ?

I appreciate any help on this.

Cheers
Nart

david-cermak, 2025-02-11:
Hi @nart-schinackow-munoz 

I'm sorry this looks like a regression, I'll fix it and add this example to CI, so we don't break it again.

Meantime, could you please check if you observe the same issue if you enable the PPP (or lwip bridge)?
```
CONFIG_PPP_SUPPORT=y
```
(these configs reserve `netif->state` for their internal data, so esp_netif won't reuse it and I think this is causing the trouble)

> My Idea is to use the esp32 connected to a Linux pc via Serial using the "slattach -v -L -s 115200 -p slip /dev/ttyUSB1"

Note that you can also use PPP netif for the same thing and it's available directly in ESP-IDF -- just choose any IDF example that uses `common_connect` component (basically all protocol examples) and choose PPP option instead of Ethernet or WiFi.

see https://github.com/espressif/esp-idf/tree/master/examples/common_components/protocol_examples_common#ppp

nart-schinackow-munoz, 2025-02-11:
Hi @david-cermak 

Thank you for your reply.

I have tried with CONFIG_PPP_SUPPORT yesterday, it works and it does not crash.
I apologise I forgot to mention this in my debug & findings.

One thing I have noticed though is when pinging from my pc the IP of the esp32 as the example suggests 
I do not get a reply for the ping even though while watching ifconfig
Tx / RX packets are counting for each ping.
Also there are no errors 
So wondering what is going on there ? Is this expected?

Many thanks

Kind regards,
Nart

david-cermak, 2025-02-11:
Did you set the IP's on linux with:
```
ifconfig sl0 $ADDR dstaddr $DEST
```
note that SLIP interface is really a simple protocol with no IP negotiation, so you need to set it up manually on your host system.
(unlike PPP, which includes IP setup and informs both parties)

nart-schinackow-munoz, 2025-02-11:
Excellent I managed to get it working, I messed up my interfaces.
I can ping now without issues and can run the example properly.
Summary of configs enabled:
CONFIG_PPP_SUPPORT 
this will fix the crash...
I look forward for the fix from your side 

Cheers
Nart

david-cermak, 2025-02-11:
I'm afraid I'll have to fix this issue in ESP-IDF.

Here I'll probably go with the workaround ^^, adding `CONFIG_PPP_SUPPORT=y` to the project's settings:

https://github.com/espressif/esp-protocols/blob/ee2fbbbee7eb4b277058c573eaa281728d4402ea/examples/esp_netif/slip_custom_netif/sdkconfig.defaults#L1-L2

closed 2025-02-20 by david-cermak