espressif/esp-protocols #765: PPP as a (SERVER) to provide IP to the opposite side (Client)

nart-schinackow-munoz, 2025-02-13:
### Answers checklist.

- [x] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [x] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

The scenario is as follows:
I have a linux PC running pppd daemon as follows:
sudo pppd /dev/ttyUSB1 115200 noauth local defaultroute nodetach debug
Via UART I have an ESP32S3 connected to the UART port on the other side running PPP server (CONFIG_LWIP_PPP_SERVER_SUPPORT enabled)

below is a snippet of what i have in terms of ppp_config
```
  ppp_config.ppp_error_event_enabled = true;
  ppp_config.ppp_phase_event_enabled = true;
#ifdef CONFIG_LWIP_PPP_SERVER_SUPPORT
    ppp_config.ppp_our_ip4_addr.addr = ESP_IP4TOADDR(192, 168, 2, 1);
    ppp_config.ppp_their_ip4_addr.addr = ESP_IP4TOADDR(192, 168, 2, 2);
#endif
```

this scenario works only if I run pppd as follows
sudo pppd /dev/ttyUSB1 115200 noauth local 192.168.1.86:192.168.1.87 defaultroute nodetach debug ms-dns 8.8.8.8 ms-dns 8.8.4.4
which in this case the pppd daemon supplies the IP

I have done some digging and debugging and managed to find a solution for this as follows:
by modifying esp_netif_lwip_ppp.c function esp_err_t esp_netif_start_ppp(esp_netif_t *esp_netif)

```
#ifdef CONFIG_LWIP_PPP_SERVER_SUPPORT
    if (ppp_ctx->ppp_our_ip4_addr.addr != IPADDR_ANY) {
        // Set our preferred address, and accept the remote
        ppp_ctx->ppp->ipcp_wantoptions.ouraddr = ppp_ctx->ppp_our_ip4_addr.addr;
        ppp_ctx->ppp->ipcp_wantoptions.accept_remote = 1;
        **ppp_set_ipcp_ouraddr(ppp_ctx->ppp, &ppp_ctx->ppp_our_ip4_addr);//nsm remove this line to go to original**
    }
    if (ppp_ctx->ppp_their_ip4_addr.addr != IPADDR_ANY) {
        // Set their preferred address, and accept the local
        ppp_ctx->ppp->ipcp_wantoptions.hisaddr = ppp_ctx->ppp_their_ip4_addr.addr;
        ppp_ctx->ppp->ipcp_wantoptions.accept_local = 1;
        **ppp_set_ipcp_hisaddr(ppp_ctx->ppp, &ppp_ctx->ppp_their_ip4_addr);//nsm remove this line to go to original**
    }
#endif // CONFIG_LWIP_PPP_SERVER_SUPPORT

```

Please see above in bold 
ppp_set_ipcp_ouraddr end up setting ppp->ask_for_local which in turn will send the IP to the client.

is there any clean way to achieve this without me modifying the actual code and keep this as minimal as possible ?

Many thanks

Kind regards,
Nart

david-cermak, 2025-02-14:
Hi Nart,
this should work.

ESP32 side:
```cpp
    ....  // use default config
    netif = esp_netif_new(&netif_ppp_config);
    ESP_ERROR_CHECK(esp_netif_ppp_get_params(netif, &netif_params));
    netif_params.ppp_our_ip4_addr.addr = ESP_IP4TOADDR(192, 168, 2, 1);
    netif_params.ppp_their_ip4_addr.addr = ESP_IP4TOADDR(192, 168, 2, 2);
    ESP_ERROR_CHECK(esp_netif_ppp_set_params(netif, &netif_params));
```

pppd side:
```
pppd /dev/ttyUSB1 115200 : modem local noauth debug nocrtscts nodetach +ipv6
```

nart-schinackow-munoz, 2025-02-14:
Hi David,

I ran this:
`pppd /dev/ttyUSB1 115200 : modem local noauth debug nocrtscts nodetach +ipv6`

Below is my ppp_config set-up
```
   esp_netif_ppp_config_t ppp_config;
    ESP_ERROR_CHECK(esp_netif_ppp_get_params(esp_netif, &ppp_config));
    ppp_config.ppp_error_event_enabled = true;
    ppp_config.ppp_phase_event_enabled = true;

#ifdef CONFIG_LWIP_PPP_SERVER_SUPPORT
    ppp_config.ppp_our_ip4_addr.addr = ESP_IP4TOADDR(192, 168, 2, 1);
    ppp_config.ppp_their_ip4_addr.addr = ESP_IP4TOADDR(192, 168, 2, 2);
#endif
```

What I get in the pppd side below

```
using channel 6
Using interface ppp0
Connect: ppp0 <--> /dev/ttyUSB1
sent [LCP ConfReq id=0x1 <asyncmap 0x0> <magic 0x9f5618bb> <pcomp> <accomp>]
rcvd [LCP ConfAck id=0x1 <asyncmap 0x0> <magic 0x9f5618bb> <pcomp> <accomp>]
sent [LCP ConfReq id=0x1 <asyncmap 0x0> <magic 0x9f5618bb> <pcomp> <accomp>]
rcvd [LCP ConfAck id=0x1 <asyncmap 0x0> <magic 0x9f5618bb> <pcomp> <accomp>]
rcvd [LCP ConfReq id=0x1 <asyncmap 0x0> <magic 0x586743a> <pcomp> <accomp>]
sent [LCP ConfAck id=0x1 <asyncmap 0x0> <magic 0x586743a> <pcomp> <accomp>]
sent [LCP EchoReq id=0x0 magic=0x9f5618bb]
sent [CCP ConfReq id=0x1 <deflate 15> <deflate(old#) 15> <bsd v1 15>]
sent [IPCP ConfReq id=0x1 <compress VJ 0f 01> <addr 0.0.0.0>]
sent [IPV6CP ConfReq id=0x1 <addr fe80::c4f5:276d:a64d:9fe3>]
rcvd [LCP EchoReq id=0x0 magic=0x586743a]
sent [LCP EchoRep id=0x0 magic=0x9f5618bb]
rcvd [IPCP ConfReq id=0x1 <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
sent [IPCP ConfRej id=0x1 <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
rcvd [IPV6CP ConfReq id=0x1 <addr fe80::4401:11ef:d663:3b6a>]
sent [IPV6CP ConfAck id=0x1 <addr fe80::4401:11ef:d663:3b6a>]
rcvd [LCP EchoRep id=0x0 magic=0x586743a]
rcvd [LCP ProtRej id=0x2 80 fd 01 01 00 0f 1a 04 78 00 18 04 78 00 15 03 2f]
Protocol-Reject for 'Compression Control Protocol' (0x80fd) received
rcvd [IPCP ConfRej id=0x1 <compress VJ 0f 01>]
sent [IPCP ConfReq id=0x2 <addr 0.0.0.0>]
rcvd [IPV6CP ConfAck id=0x1 <addr fe80::c4f5:276d:a64d:9fe3>]
local  LL address fe80::c4f5:276d:a64d:9fe3
remote LL address fe80::4401:11ef:d663:3b6a
Script /etc/ppp/ipv6-up started (pid 33537)
Script /etc/ppp/ipv6-up finished (pid 33537), status = 0x0
rcvd [IPCP ConfReq id=0x2 <addrs 0.0.0.0 192.168.2.2>]
sent [IPCP ConfRej id=0x2 <addrs 0.0.0.0 192.168.2.2>]
rcvd [IPCP ConfNak id=0x2 <addr 192.168.2.2>]
sent [IPCP ConfReq id=0x3 <addr 192.168.2.2>]
rcvd [IPCP ConfReq id=0x3]
sent [IPCP ConfAck id=0x3]
rcvd [IPCP ConfAck id=0x3 <addr 192.168.2.2>]
Could not determine remote IP address: defaulting to 10.64.64.64
Script /etc/ppp/ip-pre-up started (pid 33539)
Script /etc/ppp/ip-pre-up finished (pid 33539), status = 0x0
local  IP address 192.168.2.2
remote IP address 10.64.64.64
Script /etc/ppp/ip-up started (pid 33542)
Script /etc/ppp/ip-up finished (pid 33542), status = 0x0
rcvd [IPCP TermReq id=0x4 "Refused our IP address"]
IPCP terminated by peer (Refused our IP address)
Connect time 0.0 minutes.
Sent 0 bytes, received 26 bytes.
Script /etc/ppp/ip-down started (pid 33546)
sent [IPCP TermAck id=0x4]
Script /etc/ppp/ip-down finished (pid 33546), status = 0x0
```
**Notice the remote IP is 10.64.64.64**

ifconfig 

```
ppp0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet6 fe80::c4f5:276d:a64d:9fe3  prefixlen 128  scopeid 0x20<link>
        ppp  txqueuelen 3  (Point-to-Point Protocol)
        RX packets 12  bytes 292 (292.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 15  bytes 363 (363.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```
no IP is supplied... 
If I change esp_netif_lwip_ppp.c function esp_err_t esp_netif_start_ppp(esp_netif_t *esp_netif)

#ifdef CONFIG_LWIP_PPP_SERVER_SUPPORT
    if (ppp_ctx->ppp_our_ip4_addr.addr != IPADDR_ANY) {
        // Set our preferred address, and accept the remote
        ppp_ctx->ppp->ipcp_wantoptions.ouraddr = ppp_ctx->ppp_our_ip4_addr.addr;
        ppp_ctx->ppp->ipcp_wantoptions.accept_remote = 1;
        ppp_set_ipcp_ouraddr(ppp_ctx->ppp, &ppp_ctx->ppp_our_ip4_addr);//nsm remove this line to go to original
    }
    if (ppp_ctx->ppp_their_ip4_addr.addr != IPADDR_ANY) {
        // Set their preferred address, and accept the local
        ppp_ctx->ppp->ipcp_wantoptions.hisaddr = ppp_ctx->ppp_their_ip4_addr.addr;
        ppp_ctx->ppp->ipcp_wantoptions.accept_local = 1;
        ppp_set_ipcp_hisaddr(ppp_ctx->ppp, &ppp_ctx->ppp_their_ip4_addr);//nsm remove this line to go to original
    }

The IP shows up as expected and gets delivered.

FYI I am using ESP-IDF: v5.3.2 

Is this something that is different on the latest ?

Many thanks

Kind regards,
Nart

david-cermak, 2025-02-14:
> FYI I am using ESP-IDF: v5.3.2

you're right, it doesn't work on `v5.3.2` for some reason, I'll look into it and fix it. sorry for the inconvenience.

david-cermak, 2025-02-14:
need to backport [99b0a76](https://github.com/espressif/esp-idf/commit/99b0a765a8da94e9d17367911bbc37942dd4df1a) to `release/v5.3`

nart-schinackow-munoz, 2025-02-14:
Thanks David, my suggestion mentioned in previous post sets "ask_for_local" to 1 when using 
ppp_set_ipcp 

Cheers
Nart