espressif/esp-protocols #778: websocket have a concept failure in taking and releasing the semaphores

filzek, 2025-03-10:
### Answers checklist.

- [x] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [x] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The websocket task currently have a discrepance in the holding the sempaphore.

It just need to acquire the sempahore in the socker writing part and release it immediately so, any other tasks or even the data processing in the event callback should be enought to run without a locking and never acquiring the sempahore.

The code is not quite correct when data need to be processed inside a nested lock from callback and queues, so dead lock can occurs to the timeout without being able to take the semaphore.

The client websocket concept is good but has this failure that need to be overcome.

Maybe the correct action was to use a queue to send all data to it as pointers allocated memory so the websocket send and process it sequentially without any problem or address the streaming and concurrence in the semaphore taken.

filzek, 2025-03-10:
the locked can occurs if data is send and received and pingpong frames rans, so, there is umpredictable actions in the current design.

Event loop must be pinned to a different core of websocket task to be able to run it in full duplex.

the sending code should acquire the semaphore ONLY in the exact moment of sending, if fail, should proceed.

The websocket client design need to be changed to match the realist usage in a simple and complex scenario. The simple approach shall remove recursive and only take the semaphore if it needs to send data to the socket. another consideration is that is the data is being send, the ping/pong frame should be market as OK until next time be match, so no need to have a time wait to acquire the semaphore, should be 0 and the operation should clear the ping/pong.

The appraisal of esp_websocket_client_set_headers is an unclear action as there is no place it is see to be called in the websocket design and the timewait is eternal as:
xSemaphoreTakeRecursive(client->lock, portMAX_DELAY);, 
so, it will try to take the semaphore forever, as this is a user action and now the websocket design, this need to be clarified to the user.

similar to this

if (_tick_get_ms() - client->ping_tick_ms > client->config->ping_interval_sec * 1000) {
                     client->ping_tick_ms = _tick_get_ms();
                     ESP_LOGD(TAG, "Sending PING...");
                     if (xSemaphoreTakeRecursive(client->lock, pdMS_TO_TICKS(100)) == pdPASS) {
                         esp_transport_ws_send_raw(client->transport, WS_TRANSPORT_OPCODES_PING | WS_TRANSPORT_OPCODES_FIN, NULL, 0, client->config->network_timeout_ms);
                         xSemaphoreGiveRecursive(client->lock);
                         if (!client->wait_for_pong_resp && client->config->pingpong_timeout_sec) {
                            client->pingpong_tick_ms = _tick_get_ms();
                            client->wait_for_pong_resp = true;
                        }
                     }
                     else {
                        if (condition_that_webscoket_is_sending_data){
                            client->pingpong_tick_ms = _tick_get_ms();
                            client->wait_for_pong_resp = false;    
                        }
                        
                     }
                     
                 }

so the semaphore shall be taken exactly when need and if fail and the data is being send elsewhere, it should than bypass it.
its a not flawless design but should eliminate the deadlocks for ping/pongs frames.

for sure ping/pong frames should neved takes more than even 50ms to acquire any lock, or the message system need to be modified to a stack pile queue for ping;/pong rather sequentialy tasking.

filzek, 2025-03-10:
We have changed the semaphore so that it is only activated when needed. This update addresses the issue that was occurring frequently, and its behavior is now improved.

filzek, 2025-04-01:
We did not subit it yet as it cause some other failures that we arent yet able to solve in the current design such as:
```
======================== THREADS INFO =========================
#0  remove_free_block (control=0x3ffe4364, block=0x3fff65dc, fl=10, sl=3) at C:/Espressif/frameworks/esp-idf-v5.4/components/heap/tlsf/tlsf_control_functions.h:375
#1  block_locate_free (control=0x3ffe4364, size=<synthetic pointer>) at C:/Espressif/frameworks/esp-idf-v5.4/components/heap/tlsf/tlsf_control_functions.h:619
#2  tlsf_malloc (tlsf=0x3ffe4364, size=<optimized out>) at C:/Espressif/frameworks/esp-idf-v5.4/components/heap/tlsf/tlsf.c:443
#3  0x4009c336 in multi_heap_malloc_impl (heap=0x3ffe4350, size=1784) at C:/Espressif/frameworks/esp-idf-v5.4/components/heap/multi_heap.c:209
#4  0x40083078 in aligned_or_unaligned_alloc (offset=0, heap=<optimized out>, size=<optimized out>, alignment=<optimized out>) at C:/Espressif/frameworks/esp-idf-v5.4/components/heap/heap_caps_base.c:82
#5  heap_caps_aligned_alloc_base (alignment=<optimized out>, size=<optimized out>, caps=<optimized out>) at C:/Espressif/frameworks/esp-idf-v5.4/components/heap/heap_caps_base.c:156
#6  0x4008309c in heap_caps_malloc_base (size=1784, caps=6144) at C:/Espressif/frameworks/esp-idf-v5.4/components/heap/heap_caps_base.c:176
#7  0x40082c1e in heap_caps_malloc_default (size=1784) at C:/Espressif/frameworks/esp-idf-v5.4/components/heap/heap_caps.c:118
#8  0x4009d760 in _calloc_r (r=<optimized out>, nmemb=1, size=<optimized out>) at C:/Espressif/frameworks/esp-idf-v5.4/components/newlib/heap.c:65
#9  calloc (n=1, size=<optimized out>) at C:/Espressif/frameworks/esp-idf-v5.4/components/newlib/heap.c:29
#10 0x40198a13 in esp_tls_init () at C:/Espressif/frameworks/esp-idf-v5.4/components/esp-tls/esp_tls.c:170
#11 0x4019ad8f in ssl_connect (t=0x3fff3b94, host=0x3fff3a5c <error: Cannot access memory at address 0x3fff3a5c>, port=443, timeout_ms=5000) at C:/Espressif/frameworks/esp-idf-v5.4/components/tcp_transport/transport_ssl.c:111
#12 0x4020a8d4 in esp_transport_connect (t=0x3fff3b94, host=0x3fff3a5c <error: Cannot access memory at address 0x3fff3a5c>, port=443, timeout_ms=5000) at C:/Espressif/frameworks/esp-idf-v5.4/components/tcp_transport/transport.c:123
#13 0x4019b629 in ws_connect (t=0x3fff3c7c, host=0x3fff3a5c <error: Cannot access memory at address 0x3fff3a5c>, port=443, timeout_ms=5000) at C:/Espressif/frameworks/esp-idf-v5.4/components/tcp_transport/transport_ws.c:201
#14 0x4020a8d4 in esp_transport_connect (t=0x3fff3c7c, host=0x3fff3a5c <error: Cannot access memory at address 0x3fff3a5c>, port=443, timeout_ms=5000) at C:/Espressif/frameworks/esp-idf-v5.4/components/tcp_transport/transport.c:123
#15 0x40130606 in esp_websocket_client_task (pv=0x3fff3798) at D:/Dropbox/Dev/WiseHome/managed_components/espressif__esp_websocket_client/esp_websocket_client.c:1025
#16 0x40099ab1 in vPortTaskWrapper (pxCode=0x40130530 <esp_websocket_client_task>, pvParameters=0x3fff3798) at C:/Espressif/frameworks/esp-idf-v5.4/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
Retrying reading threads information...
```

we can send the websocket code with the semaphore moved outside the while to your appreciation as well, it is attached.

but until we can solve this problem above is not the best way to use it yet.

filzek, 2025-04-02:
it seens that the client->config->host is freed somewhere among the usage and the connection could not be set, not sure why, as the task and the handler could be doing two separate things and was not started yet, seens something like concurrence, so we are adding the semaphore where it is needed to be taken to make sure no lock or change occur in a place that must be checked prior, but this is not the best act to solve the problem.

glmfe, 2025-07-01:
Hello @filzek, I'm coming back to this ticket, do you have any update to share?

glmfe, 2025-07-01:
that's great news, I think we can close this issue then. If you confirm these other issues are from websocket then you can open a new one, does that sound good to you?

glmfe, 2025-07-01:
sure, let's wait then :)

filzek, 2025-08-27:
1.5.0 seems quite stable now, we are testing and there is issues only when transport layer falls

closed 2025-08-28 by glmfe