espressif/esp-protocols #787: In pppos_example, esp_modem_get_signal_quality failed with 263 ESP_ERR_TIMEOUT

as4mz, 2025-03-21:
### Answers checklist.

- [x] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [x] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am getting the following error while I am trying to run the PPPoS simple client example:
`pppos_example: esp_modem_get_signal_quality failed with 263 ESP_ERR_TIMEOUT`

My IDF version is: 5.4
esp_model version is: 1.3.0

I am using SIM7600G-H modem: 
https://www.waveshare.com/product/sim7600g-h-4g-dongle.htm

I am using pins 15 and 16 for RXD and TXD, respectively; pins 14 and 15 for CTS and RTS, respectively.

as4mz, 2025-03-21:
I have also tried to increse the timeout for RTC watchdog but it has not been successful.

david-cermak, 2025-03-21:
> Also, trying iner-changing the RXD and TXD pins and got the following error:

this seems to have fixed the initial connection issues, but you're failing to set the PDP context or switching to PPP mode. Please double check with your provider that the APN is correct.

as4mz, 2025-03-23:
According to T-mobile, the APN should be `fast.t-mobile.com`

https://www.t-mobile.com/support/devices/not-sold-by-t-mobile/byod-t-mobile-data-and-apn-settings

Is it related to my SIM7600 hardware? Do you suggest any other vendors for modem hardware?

as4mz, 2025-03-25:
Here are the logs:

`
--- Warning: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM8 instead...
--- esp-idf-monitor 1.5.0 on \\.\COM8 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
I (33) boot: ESP-IDF v5.4 2nd stage bootloader
I (33) boot: compile time Mar 25 2025 11:32:48
I (33) boot: Multicore bootloader
I (33) boot: chip revision: v0.1
I (36) boot: efuse block revision: v1.2
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : SLOW READ
I (48) boot.esp32s3: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3c070020 size=20c6ch (134252) map
I (127) esp_image: segment 1: paddr=00030c94 vaddr=3fc94100 size=02cd8h ( 11480) load
I (130) esp_image: segment 2: paddr=00033974 vaddr=40374000 size=0c6a4h ( 50852) load
I (145) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=6eea8h (454312) map
I (251) esp_image: segment 4: paddr=000aeed0 vaddr=403806a4 size=03a30h ( 14896) load
I (256) esp_image: segment 5: paddr=000b2908 vaddr=600fe100 size=0001ch (    28) load
I (262) boot: Loaded app from partition at offset 0x10000
I (262) boot: Disabling RNG early entropy source...
V (277) mmap: after coalescing, 1 regions are left
I (277) cpu_start: Multicore app
D (277) cpu_start: Pro cpu up
D (277) cpu_start: Starting app cpu, entry point is 0x403753e4
--- 0x403753e4: call_start_cpu1 at C:/esp/esp-idf-v5.4/components/esp_system/port/cpu_start.c:196

D (0) cpu_start: App cpu up
V CACHE_ERR: illegal error intr clr & ena mask is: 0x3f
V CACHEDE(2R:)cork: 1 access CLrror inrar clr & ena mask 10
 0x1f
V CACHE_ERR: illegal error intr clr & ena mask is: 0x3f
V CACHE_ERR: core 0 access error intr clr & ena mask is: 0x1f
D (307) cpu_start: calling init function: 0x42003070 on core: 0
--- 0x42003070: __esp_system_init_fn_init_efuse_check at C:/esp/esp-idf-v5.4/components/efuse/src/esp_efuse_startup.c:36

D (311) cpu_start: calling init function: 0x420034c8 on core: 0
--- 0x420034c8: __esp_system_init_fn_init_show_cpu_freq at C:/esp/esp-idf-v5.4/components/esp_system/startup_funcs.c:58

I (316) cpu_start: Pro cpu start user code
I (320) cpu_start: cpu freq: 160000000 Hz
D (324) cpu_start: calling init function: 0x42002fa0 on core: 0
--- 0x42002fa0: __esp_system_init_fn_init_show_app_info at C:/esp/esp-idf-v5.4/components/esp_app_format/esp_app_desc.c:123

I (330) app_init: Application information:
I (333) app_init: Project name:     pppos_client
I (338) app_init: App version:      1
I (341) app_init: Compile time:     Mar 25 2025 11:32:25
I (346) app_init: ELF file SHA256:  9d47df626...
[... 449 lines cut ...]
I (955) main_task: Calling app_main()
D (955) esp_netif_lwip: LwIP stack has been initialized
D (965) esp_netif_lwip: esp-netif has been successfully initialized
D (965) event: running task for loop 0x3fc9dbb0
D (975) event: created task for loop 0x3fc9dbb0
D (975) event: created event loop 0x3fc9dbb0
D (975) esp_netif_lwip: check: remote, if=0x3fc9b0b0 fn=0x4201dc80
--- 0x4201dc80: esp_netif_new_api at C:/esp/esp-idf-v5.4/components/esp_netif/lwip/esp_netif_lwip.c:712

V (985) esp_netif_objects: esp_netif_add_to_list_unsafe 0x3fc9eb60
D (995) esp_netif_objects: esp_netif_add_to_list_unsafe netif added successfully (total netifs: 1)
D (1005) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3fc9eda0
D (1005) esp-netif_lwip-ppp: Phase Dead
D (1015) esp_netif_lwip: call api in lwip: ret=0x0, give sem
I (1015) uart: queue free spaces: 30
V (1015) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1025) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (1035) intr_alloc: Connected src 28 to int 8 (cpu 0)
I (1035) pppos_example: Initializing esp_modem for the SIM7600 module...
V (1045) command_lib: get_signal_quality
V (1045) command_lib: generic_get_string
D (1055) uart-tx: 0x3fc9b04c   41 54 2b 43 53 51 0d                              |AT+CSQ.|
E (1565) pppos_example: esp_modem_get_signal_quality failed with 263 ESP_ERR_TIMEOUT
I (1565) main_task: Returned from app_main()
`

as4mz, 2025-03-25:
I have also tried resetting the modem before checking the singal quality as this link suggested: https://github.com/espressif/esp-protocols/issues/232

However, I am able to send messages to the my number sometimes. I am not sure when extactly this happens.

as4mz, 2025-03-25:
I have also tried to get the IMEI and send AT command using: `esp_modem_get_imei `and `esp_modem_at`. None is working. Is there any connection issue?

as4mz, 2025-03-31:
@david-cermak Can you help me to figure out the issue? Am I missing something on hardware connections (pin-layout)? 

Thanks for your help

david-cermak, 2025-03-31:
@as4mz Are you running the pppos client example as is, without any changes? Could you please list the changes and the configuration? Any chance you're using some HW or SW flow control?

```
D (1050) uart-tx: 0x3fc9b04c   41 54 2b 43 53 51 0d                              |AT+CSQ.|
D (1060) uart-rx: 0x3fca191c   00                                                |.|
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).
```
This really looks strange that the WDT triggers in interrupt (in UART?), could you please check if moving UART ISR to IRAM (Kconfig option) would help?

as4mz, 2025-04-02:
@david-cermak thanks for your directions. I have removed and downloaded the example again. This time I mainly changed the Pin-layout in the config file. No flow controlling or anything else. 

This is the list of changes:

```
Place UART ISR function into IRAM: True
TXD Pin Number- 5
RXD Pin Number- 4
CTS Pin Number- 1
RTS Pin Number- 2
Set preferred modem control flow: None
```

I got this logs then:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x42067f63
--- 0x42067f63: uart_ll_get_txfifo_len at C:/esp/esp-idf-v5.4/components/hal/esp32s3/include/hal/uart_ll.h:410
 (inlined by) uart_tx_char at C:/esp/esp-idf-v5.4/components/esp_driver_uart/src/uart_vfs.c:170

SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce2810,len:0x216c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xeb8
load:0x403cb700,len:0x3400
entry 0x403c8970
I (36) boot: ESP-IDF v5.4 2nd stage bootloader
I (36) boot: compile time Apr  2 2025 10:10:09
I (36) boot: Multicore bootloader
D (37) bootloader_flash: non-XMC chip detected by SFDP Read (87), skip.
D (42) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (49) boot: chip revision: v0.1
I (52) boot: efuse block revision: v1.2
D (55) boot.esp32s3: magic e9
D (58) boot.esp32s3: segments 04
D (61) boot.esp32s3: spi_mode 02
D (64) boot.esp32s3: spi_speed 0f
D (67) boot.esp32s3: spi_size 01
I (70) boot.esp32s3: Boot SPI Speed : 80MHz
I (74) boot.esp32s3: SPI Mode       : SLOW READ
I (78) boot.esp32s3: SPI Flash Size : 2MB
D (82) boot: Enabling RTCWDT(9000 ms)
I (85) boot: Enabling RNG early entropy source...
D (90) bootloader_flash: rodata starts from paddr=0x00008000, size=0xc00, will be mapped to vaddr=0x3c000000
V (99) bootloader_flash: after mapping, starting from paddr=0x00000000 and vaddr=0x3c000000, 0x10000 bytes are mapped
D (109) boot: mapped partition table 0x8000 at 0x3c008000
D (115) flash_parts: partition table verified, 4 entries
I (120) boot: Partition Table:
I (122) boot: ## Label            Usage          Type ST Offset   Length
D (129) boot: load partition table entry 0x3c008000
D (133) boot: type=1 subtype=2
I (136) boot:  0 nvs              WiFi data        01 02 00009000 00006000
D (143) boot: load partition table entry 0x3c008020
D (147) boot: type=1 subtype=1
I (150) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (157) boot: load partition table entry 0x3c008040
D (161) boot: type=0 subtype=0
I (164) boot:  2 factory          factory app      00 00 00010000 00100000
I (171) boot: End of partition table
D (174) boot: Trying partition index -1 offs 0x10000 size 0x100000
D (180) esp_image: reading image header @ 0x10000
D (184) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (191) esp_image: image header: 0xe9 0x06 0x02 0x01 40375550
V (196) esp_image: loading segment header 0 at offset 0x10018
[... 266 lines cut ...]
D (1461) esp_netif_lwip: LwIP stack has been initialized
D (1471) esp_netif_lwip: esp-netif has been successfully initialized
D (1471) event: running task for loop 0x3fc9e8b0
D (1481) event: created task for loop 0x3fc9e8b0
D (1481) event: created event loop 0x3fc9e8b0
D (1491) esp_netif_lwip: check: remote, if=0x3fc9bdb0 fn=0x4201a7b4
--- 0x4201a7b4: esp_netif_new_api at C:/esp/esp-idf-v5.4/components/esp_netif/lwip/esp_netif_lwip.c:712

V (1491) esp_netif_objects: esp_netif_add_to_list_unsafe 0x3fc9f860
D (1501) esp_netif_objects: esp_netif_add_to_list_unsafe netif added successfully (total netifs: 1)
D (1511) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3fc9faa0
D (1511) esp-netif_lwip-ppp: Phase Dead
D (1521) esp_netif_lwip: call api in lwip: ret=0x0, give sem
I (1521) pppos_example: Initializing esp_modem for the SIM7600 module...
I (1531) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (1541) uart: queue free spaces: 30
V (1541) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1551) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E
D (1561) intr_alloc: Connected src 28 to int 8 (cpu 0)
V (1561) command_lib: get_signal_quality
V (1561) command_lib: generic_get_string
D (1571) uart-tx: 0x3fc9bd4c   41 54 2b 43 53 51 0d                              |AT+CSQ.|
E (2081) pppos_example: esp_modem_get_signal_quality failed with 263 ESP_ERR_TIMEOUT
I (2081) main_task: Returned from app_main()
```

as4mz, 2025-04-02:
I see I got this error immediately: `ESP_ERR_TIMEOUT`
Usually, it should wait for the modem to trying to establish the connection, right?

as4mz, 2025-04-07:
@kumekay Do you know why I am getting this `Phase Dead` error Just immediately after the connection estabilshed?
```
D (1511) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3fc9faa0
D (1511) esp-netif_lwip-ppp: Phase Dead
```

GuillaumeHupi, 2025-04-23:
Hello, 

I have exactly the same issue with a slightly different configuration : 

I use ESP-IDF v5.3.0 (I also tried with 5.4.0 and got the same result).
I use the last version of esp-protocols (downloaded it yesterday). 

I use a t-sim-cam V1.3 from lilygo https://lilygo.cc/products/t-simcam?srsltid=AfmBOooO_hVRouLhYwN7xXos0f-D70-CiQXjw2pWNpCSpw-PSqZ0RITD with a SIM7070G modem in a pcie format https://lilygo.cc/products/a-t-pcie?variant=42335921963189. 

On this board, the RTS and CTS pins of the modem are not wired so I don't use them and I only use TX and RX pins without flow control. 

So first, I am able to use the modem with this board with AT commands, this part works fine. 
Secondly, I am able to use the modem with this board with pppos, but with the arduino library https://github.com/espressif/arduino-esp32/blob/master/libraries/PPP/examples/PPP_Basic/PPP_Basic.ino. After configuring the example for my board like this : 

```C
#include <PPP.h>

#define PPP_MODEM_APN "iot.1nce.net"
#define PPP_MODEM_PIN NULL  // or NULL

// WaveShare SIM7600 HW Flow Control
// #define PPP_MODEM_RST       25
// #define PPP_MODEM_RST_LOW   false  //active HIGH
// #define PPP_MODEM_RST_DELAY 200
// #define PPP_MODEM_TX        21
// #define PPP_MODEM_RX        22
// #define PPP_MODEM_RTS       26
// #define PPP_MODEM_CTS       27
// #define PPP_MODEM_FC        ESP_MODEM_FLOW_CONTROL_HW
// #define PPP_MODEM_MODEL     PPP_MODEM_SIM7600

// SIM800 basic module with just TX,RX and RST
#define PPP_MODEM_RST     48
#define PPP_MODEM_RST_LOW false //active LOW
#define PPP_MODEM_TX      45
#define PPP_MODEM_RX      46
#define PPP_MODEM_RTS     -1
#define PPP_MODEM_CTS     -1
#define PPP_MODEM_FC      ESP_MODEM_FLOW_CONTROL_NONE
#define PPP_MODEM_MODEL   PPP_MODEM_SIM7070
#define PPP_MODEM_RST_DELAY 2000

#define PWR_ON_PIN 1

void onEvent(arduino_event_id_t event, arduino_event_info_t info) {
  switch (event) {
    case ARDUINO_EVENT_PPP_START:        Serial.println("PPP Started"); break;
    case ARDUINO_EVENT_PPP_CONNECTED:    Serial.println("PPP Connected"); break;
    case ARDUINO_EVENT_PPP_GOT_IP:       Serial.println("PPP Got IP"); break;
    case ARDUINO_EVENT_PPP_LOST_IP:      Serial.println("PPP Lost IP"); break;
    case ARDUINO_EVENT_PPP_DISCONNECTED: Serial.println("PPP Disconnected"); break;
    case ARDUINO_EVENT_PPP_STOP:         Serial.println("PPP Stopped"); break;
    default:                             break;
  }
}

void testClient(const char *host, uint16_t port) {
  NetworkClient client;
  if (!client.connect(host, port)) {
    Serial.println("Connection Failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    client.read();  //Serial.write(client.read());
  }

  Serial.println("Connection Success");
  client.stop();
[... 2344 lines cut ...]
    gpio_set_level(MODEM_RST_KEY, 1);
    vTaskDelay(pdMS_TO_TICKS(2000));
    gpio_set_level(MODEM_RST_KEY, 0);
    vTaskDelay(pdMS_TO_TICKS(5000));
```

And I got exactly the same behaviour than @as4mz which is : 

```
D (7765) esp_netif_lwip: check: remote, if=0x3fc9b270 fn=0x4201ced0
--- 0x4201ced0: esp_netif_new_api at /Users/guillaumecostes/esp/v5.3/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:690

D (7765) esp_netif_objects: esp_netif_add_to_list_unsafe netif added successfully (total netifs: 1)
D (7765) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3fc9ef90
D (7775) esp-netif_lwip-ppp: Phase Dead
D (7775) esp_netif_lwip: call api in lwip: ret=0x0, give sem
I (7785) pppos_example: Initializing esp_modem for the SIM7070 module...
I (7795) uart: queue free spaces: 30
D (7795) intr_alloc: Connected src 28 to int 8 (cpu 0)
D (7805) uart-tx: 0x3fc9b20c   41 54 2b 43 53 51 0d                              |AT+CSQ.|
E (8305) pppos_example: esp_modem_get_signal_quality failed with 263 ESP_ERR_TIMEOUT
I (8305) main_task: Returned from app_main()
```

I think I am doing something wrong in the settings, since with arduino library, pppos is working, but I can't figure out what.