espressif/esp-protocols #789: SIM7600 Strange packet parsing behaviour

kgrimsby, 2025-03-26:
### Answers checklist.

- [x] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [x] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to get SIM7600 to work on Lilygo T-SIM7600. I must be doing something very very strange - but I don't know what. First I hade problems getting an IP, but by disabling CONFIG_LWIP_PPP_VJ_HEADER_COMPRESSION it worked fine.

Trying to make a simple http call though, results in error.

```
D (901) intr_alloc: Connected src 24 to int 2 (cpu 0)
D (906) app_start: Starting scheduler on CPU0
I (910D (910) intr_alloc: Connected src 25 to int 1 (cpu 1)
D (910) app_start: Starting scheduler on CPU1
) main_task: Started on CPU0
D (920) heap_init: New heap initialised at 0x3ffe0440
D (920) heap_init: New heap initialised at 0x3ffe4350
D (930) intr_alloc: Connected src 16 to int 3 (cpu 0)
I (930) main_task: Calling app_main()
D (940) lwip: dns_init: initializing

D (940) esp_netif_lwip: LwIP stack has been initialized
D (940) esp_netif_lwip: esp-netif has been successfully initialized
D (950) event: running task for loop 0x3ffb8768
D (950) event: created task for loop 0x3ffb8768
D (960) event: created event loop 0x3ffb8768
I (960) app: toggling modem power...
I (970) app: status=1
E (970) app: Modem is already powered on
I (3570) app: status=1
I (4570) app: status=1
I (5570) app: status=1
I (6570) app: status=1
I (7570) app: status=1
I (8570) app: status=1
I (9570) app: status=1
I (10570) app: status=1
I (11570) app: status=1
I (12570) app: status=1
I (13570) app: status=1
I (14570) app: status=1
I (15570) app: status=1
I (16570) app: status=1
I (17570) app: status=1
I (18570) app: status=1
I (19570) app: status=1
I (20570) app: status=1
I (21570) app: status=1
I (22570) app: status=1
I (23570) app: status=1
I (24570) app: status=0
I (25870) app: waiting for modem...
I (25970) app: status=1
D (25970) esp_netif_lwip: check: remote, if=0x3ffb6430 fn=0x400e4e40
--- 0x400e4e40: esp_netif_new_api at /Users/kristiangrimsby/esp/v5.4/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:712

D (25970) esp_netif_objects: esp_netif_add_to_list_unsafe netif added successfully (total netifs: 1)
D (25970) lwip: ppp phase changed[1]: phase=0

D (25980) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3ffb9968
D (25980) esp-netif_lwip-ppp: Phase Dead
D (25990) esp_netif_lwip: call api in lwip: ret=0x0, give sem
I (25990) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (26000) uart: queue free spaces: 50
D (26000) intr_alloc: Connected src 35 to int 5 (cpu 0)
D (26010) uart-tx: 0x3f419c28   7e ff 03 c0 21 09 01 00  08 99 d1 35 c1 8e 2c 7e  |~...!......5..,~|
D (26220) uart-tx: 0x3f419c18   0d 0a 41 54 0d 0a 00                              |..AT...|
[... 845 lines cut ...]
    int rssi, ber;
    err = esp_modem_get_signal_quality(dce, &rssi, &ber);
    if (err != ESP_OK)
    {
      ESP_LOGE(TAG, "esp_modem_get_signal_quality failed with %d %s", err, esp_err_to_name(err));
    }
    else
    {
      ESP_LOGI(TAG, "Signal quality: rssi=%d, ber=%d", rssi, ber);
    }
    */

    esp_netif_ip_info_t ip_info;
    esp_err_t err2 = esp_netif_get_ip_info(esp_netif, &ip_info);
    if (err2 == ESP_OK)
    {
      ESP_LOGE(TAG, "Connection IP: " IPSTR, IP2STR(&ip_info.ip));
    }

    http_test();
  }
}
```

What am I doing wrong here?

david-cermak, 2025-03-26:
>   dte_config.uart_config.flow_control = ESP_MODEM_FLOW_CONTROL_SW;

Can you try to turn off the flow control?

>     err = esp_modem_set_mode(dce, ESP_MODEM_MODE_CMUX);

Also would it work without the CMUX mode? switching directly to `ESP_MODEM_MODE_DATA` ?

kgrimsby, 2025-03-26:
Oh my, of course it was the flow control. About exhausted every option but missed that one.

That seems to work flawless!

closed 2025-03-26 by kgrimsby