espressif/esp-protocols #792: Extend modem_tcp_client to show how to use it with multiple protocols

KaSroka, 2025-03-27:
### Is your feature request related to a problem?

I would like to use modem_tcp_client with both mqtt and http at the same time. Current example does not show how to do this.

### Describe the solution you'd like.

modem_tcp_client shows how to use it with multiple protocols at the same time (i.e. mqtt connection is active and we do http get request at the same time)

david-cermak, 2025-04-15:
Thanks for proposing this feature request.
It's currently provided only as a demo in examples, which is not really reusable and easily extendable. If you need to add another client, you'd have to replicate the same logic for the localhost listener option (spin a new thread to handle a separate socket, which is not trivial). For the custom esp_transport option, it's a little easier, but the http-client doesn't currently support transport injection.
The only workaround I can offer at the moment is to linker-wrap IDF's esp_transport and use it for both mqtt client and http client.

We plan to support this in the upcoming `modem-v2.0` as a standard feature.

KaSroka, 2025-04-17:
Awesome, thanks for the info! When do you plan to release modem-v2.0?