espressif/esp-protocols #815: Uart warnings when re-init the modem

FabITA1, 2025-05-26:
### Answers checklist.

- [x] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [x] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm using esp_modem 1.4.0 with ESP-IDF5.4.1 on ESP32S3. 
I correctly initialize the device with esp_modem_new_dev(), then I use the modem. 
In some circumstances (communication problems or others) I need to hardware reset and restart the modem. 
To do so, I  use esp_modem_destroy(), power down/up the modem and then call a new esp_modem_new_dev(). 
When I call for the second time esp_modem_new_dev() I obtain this two warnings:
W (129801) uart: GPIO 39 is not usable, maybe used by others
W (129801) uart: GPIO 42 is not usable, maybe used by others
This warning doesn't appear the first esp_modem_new_dev() calling. 
GPIO39 & GPIO42 are RTS and CTS pin.
The modem still works correctly.

May be the uart are not completely de-init? 
What can I do to solve this problem?

david-cermak, 2025-05-27:
This seems to be handled correctly in esp_modem, the pins are allocated when creating the UART DTE -- and should be released in the destructor when calling `uart_driver_delete()` -- which should internally release all UART allocated pins.
Might be some issue in the driver, but couldn't spot any obvious issue.

Are you running `esp_modem_new_dev()` and `esp_modem_destroy()` asynchronously? does the warning go away if you add a short delay between the calls?
Is it possible to quickly can check with IDF `v5.5` or `master`? I've seen some changes in the pin reservation logic between `v5.4` and master, so maybe it's been fixed already?

FabITA1, 2025-07-23:
Hi David,
I have tried with IDF 5.5 and the problems seems to be solved. 
Sorry for late response... I was waiting for the version to be released.
Many thanks

FabITA1, 2025-12-30:
Hi @david-cermak,
in these days I upgraded the IDF to 5.5.2 and now the warning related to uart pins appears againg:
`W (18645) uart: GPIO 39 is not usable, maybe used by others`

With IDF 5.5 or 5.5.1 no warnings appears. I'm using esp_modem 2.0.0.