espressif/esp-protocols #832: The priority of esp_websocket_client_task

nopnop2002, 2025-06-27:
### Answers checklist.

- [x] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [x] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The priority of esp_websocket_client_task is fixed at 5, but shouldn't it be the priority of the calling task + 1?

```
#define WEBSOCKET_TASK_PRIORITY         (5)
```

euripedesrocha, 2025-06-27:
@nopnop2002 that is the default for the websocket client task priority. 

Priority of the task can be set using [config->task_prio](https://docs.espressif.com/projects/esp-protocols/esp_websocket_client/docs/latest/index.html#_CPPv4N29esp_websocket_client_config_t9task_prioE)

nopnop2002, 2025-06-27:
@euripedesrocha 

Thank you for comment.
OK. I will set my own prioroty.

closed 2025-06-27 by nopnop2002