espressif/esp-protocols #846: esp_modem PPP does not want to start when using cpp class based implementation

lolrobbe2, 2025-07-15:
### Answers checklist.

- [x] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [x] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

```CPP
 bool gm02s::config(std::string_view apn, int priority,
                       std::string_view dns_addr) {
        esp_err_t ret = nvs_flash_init();
        if (ret == ESP_ERR_NVS_NO_FREE_PAGES ||
            ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
            ESP_ERROR_CHECK(nvs_flash_erase());
            ret = nvs_flash_init();
        }
        
        esp_err_t netif_ret = esp_netif_init();
        if (netif_ret != ESP_OK)
            return false;

        esp_err_t netif_event_loop_ret = esp_event_loop_create_default();
        if (netif_event_loop_ret != ESP_OK)
            return false;

        ppp_event_group = xEventGroupCreate(); 
        if(!ppp_event_group){
            return false;
        }
        initReset();

        pdpContext.protocol_type = "IP";
        pdpContext.apn = apn;
        pdpContext.context_id = 1;

        if (!uartDTE && (!setupUartDTE() || !setupNetif(priority))) {
            return false;
        }

        uartDTE->set_urc_cb(callback);
        if(setupModemDCE()){
            configured = true;

            esp_event_handler_register(IP_EVENT,ESP_EVENT_ANY_ID,on_modem_event,this);
            return true;
        }
        return false;
    }
 bool gm02s::connect()
    {
        esp_modem::command_result res = getModule()->set_radio_state(0);
        if (res == esp_modem::command_result::FAIL) {
            return false;
        }
        res = getModule()->config_network_registration_urc(1);
        if (res == esp_modem::command_result::FAIL) {
          return false;
        }

        res = getModule()->set_radio_state(1);
        if (res == esp_modem::command_result::FAIL) {
            return false;
        }
        res = waitForConnection();
        if(res == esp_modem::command_result::OK){
            getModule()->set_mode(esp_modem::modem_mode::CMUX_MODE);
            ESP_LOGI("GM02S", "Waiting for network connection...");
            while (!esp_netif_is_netif_up(pppInterface)) {
              vTaskDelay(pdMS_TO_TICKS(1000)); // Wait for 1 second
            }
            ESP_LOGI("GM02S", "Network connected!");
            return true;
        }

        EventBits_t bits = xEventGroupWaitBits(
            ppp_event_group,       // The event group handle
            IP_GOT_BIT,            // The bit to wait for
            pdFALSE,               // Do NOT clear bit on exit
            pdFALSE,               // Wait for any bit
            pdMS_TO_TICKS(1000000) // TODO Timeout (10 seconds here)
        );
        return false;
    }

 void gm02s::initReset()
    {
        gpio_set_direction((gpio_num_t)hardwareConfig.pinReset, GPIO_MODE_OUTPUT);
        gpio_set_pull_mode((gpio_num_t)hardwareConfig.pinReset, GPIO_FLOATING);
        gpio_deep_sleep_hold_en();

        gpio_hold_dis((gpio_num_t)hardwareConfig.pinReset);
        gpio_set_level((gpio_num_t)hardwareConfig.pinReset, 0);
        vTaskDelay(pdMS_TO_TICKS(10));
        gpio_set_level((gpio_num_t)hardwareConfig.pinReset, 1);
        gpio_hold_en((gpio_num_t)hardwareConfig.pinReset);
        vTaskDelay(pdMS_TO_TICKS(5000));
    }

    bool gm02s::setupNetif(int priority) {
        /* the inherent config is used so we can set the route_prio parameter wich is otherwise of limits! */
        esp_netif_inherent_config_t *netif_ppp_inherent_config =
            (esp_netif_inherent_config_t *)ESP_NETIF_BASE_DEFAULT_PPP;
        netif_ppp_inherent_config->route_prio = priority;

        esp_netif_config_t netif_ppp_config = {
            .base = netif_ppp_inherent_config,
            .driver = NULL,
            .stack = ESP_NETIF_NETSTACK_DEFAULT_PPP,
        };

        pppInterface = esp_netif_new(&netif_ppp_config);
        if (!pppInterface)
            return false;

        Driver::priority = priority;
        return true;
    }

    bool gm02s::setupUartDTE()
    {
        esp_modem::dte_config dteConfig = {
            .dte_buffer_size = 4096 / 2,
            .task_stack_size = 4096,
            .task_priority = 5,
            .uart_config = {
                .port_num = hardwareConfig.uart_no,
                .data_bits = UART_DATA_8_BITS,
                .stop_bits = UART_STOP_BITS_1,
                .parity = UART_PARITY_DISABLE,
                .flow_control = ESP_MODEM_FLOW_CONTROL_HW,
                .source_clk = ESP_MODEM_DEFAULT_UART_CLK,
                .baud_rate = hardwareConfig.baudRate,
                .tx_io_num = hardwareConfig.pinTX,
                .rx_io_num = hardwareConfig.pinRX,
                .rts_io_num = hardwareConfig.pinRTS,
                .cts_io_num = hardwareConfig.pinCTS,
                .rx_buffer_size = 4096,
                .tx_buffer_size = 4096 / 2,
                .event_queue_size = 30,
            }};

        uartDTE = esp_modem::create_uart_dte(&dteConfig);
        return uartDTE != nullptr;
    }

    bool gm02s::setupModemDCE()
    {
        esp_modem::dce_config dceConfig =
            ESP_MODEM_DCE_DEFAULT_CONFIG(pdpContext.apn.data());

        gm02sDce =
            esp_modem::create_SQNGM02S_dce(&dceConfig, uartDTE, pppInterface);

        if (!gm02sDce) {
            return false;
        }

        gm02sDce->set_urc(callback);

        /* syn communication with the modem */
        for (int i = 0; i < 3; i++) {
            if (getModule()->sync() == esp_modem::command_result::OK)
                break;
            vTaskDelay(pdMS_TO_TICKS(1000));
        }

        getModule()->set_pdp_context(pdpContext);
        return true;
    }
```

lolrobbe2, 2025-07-15:
@david-cermak i don't see any PPP related code for the generic_module implementation. could this explain why esp_modem does not go into PPP mode?

closed 2025-07-15 by lolrobbe2

david-cermak, 2025-07-16:
@lolrobbe2 Please check the simplifies class diagram: https://github.com/espressif/esp-protocols/blob/master/docs/esp_modem/en/README.rst

This `generic_module` is an abstraction of module's commands and functions only, has nothing to do with network. The actual modem abstraction (DCE) is a composition of "module", "PPP" and "DTE".

The C-interface is simplified and abstracts only `DCE`.