espressif/esp-protocols #847: esp_modem: the response of dec->at_raw is truncated for cusd command

earthPerson-001, 2025-07-16:
### Answers checklist.

- [x] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [x] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

esp_modem version: 1.4.0

While performing at command `AT+CUSD=1,"*901#",15` the response is truncated.

whole command
```cpp
dce->at_raw("AT+CUSD=1,\"*903#\",15\r", str, "+CUSD", "ERROR", 10000);
ESP_LOGI(TAG, "AT response: %s", str.c_str());
```

I have tried the following
- increasing the dte buffer length
- Use inflatable buffer in DCE
- Increasing "Size in bytes for response from AT commands returning textual values (C-API)" from 128 to 256
- Increasing tx and rx buffer for uart

I also tried to piece together the responses by using dce->command function but i get various garbage values in between

Expected Output:
 `+CUSD: 0, "Your number 9802301039, Status: Active, Active date: 21/03/2022, The age of SIM 1213 days, Tariff plan: Corporate.", 15`

whole output should be like this
```txt
OK
+CUSD: 0, "Your number 9802301039, Status: Active, Active date: 21/03/2022, The age of SIM 1213 days, Tariff plan: Corporate.", 15
```
but I am trying to get the sim number so I only search for +CUSD

But the response is truncated after `Tariff plan:`

There is no newline characters in response when i query it by directly connecting to sim via usb.

```txt
D (5724) uart-tx: 0x3fcaeb74   f9 09 ef 2b                                       |...+|
D (5732) uart-tx: 0x3fcc2210   41 54 2b 43 55 53 44 3d  31 2c 22 2a 39 30 33 23  |AT+CUSD=1,"*903#|
D (5741) uart-tx: 0x3fcc2220   22 2c 31 35 0d                                    |",15.|
D (5749) uart-tx: 0x3fcaeb78   04 f9                                             |..|
D (5849) uart-rx: 0x3fcc0984   f9 09 ff 0d 0d 0a 4f 4b  0d 0a cd f9              |......OK....|
D (5850) CMUX: Payload frame: dlci:02 type:ff payload:6 available:8
D (7577) uart-rx: 0x3fcc0984   f9 09 ff ed 0d 0a 2b 43  55 53 44 3a 20 30 2c 22  |......+CUSD: 0,"|
D (7577) uart-rx: 0x3fcc0994   59 6f 75 72 20 6e 75 6d  62 65 72 20 69 73 20 39  |Your number is 9|
D (7584) uart-rx: 0x3fcc09a4   38 30 32 33 30 31 30 33  39 2c 20 53 74 61 74 75  |802301039, Statu|
D (7593) uart-rx: 0x3fcc09b4   73 3a 20 41 63 74 69 76  65 2c 20 41 63 74 69 76  |s: Active, Activ|
D (7602) CMUX: Payload frame: dlci:02 type:ff payload:118 available:60
D (7609) uart-rx: 0x3fcc09c4   65 20 64 61 74 65 3a 20  32 31 2f 30 33 2f 32 30  |e date: 21/03/20|
D (7618) uart-rx: 0x3fcc09d4   32 32 2c 20 54 68 65 20  61 67 65 20 6f 66 20 53  |22, The age of S|
D (7627) uart-rx: 0x3fcc09e4   49 4d 20 31 32 31 33 20  64 61 79 73 2c 20 54 61  |IM 1213 days, Ta|
D (7636) uart-rx: 0x3fcc09f4   72 69 66 66 20 70 6c 61  6e 3a 65 f9              |riff plan:e.|
D (7645) CMUX: Payload frame: dlci:02 type:ff payload:58 available:60
D (7652) uart-rx: 0x3fcc0984   f9 09 ff 09 20 43 6f 72  ca f9 f9 09 ff 1b 70 6f  |.... Cor......po|
D (7661) uart-rx: 0x3fcc0994   72 61 74 65 2e 22 2c 31  35 0d 0a 35 f9           |rate.",15..5.|
D (7670) CMUX: Payload frame: dlci:02 type:ff payload:4 available:25
D (7676) CMUX: Payload frame: dlci:02 type:ff payload:13 available:15
I (7683) SIM7600: AT response:
+CUSD: 0,"Your number is 9802301039, Status: Active, Active date: 21/03/2022, The age of SIM 1213 days, Tariff plan:
D (7704) command_lib: generic_command command AT+IFC=2,2

D (7709) uart-tx: 0x3fcaeb94   f9 09 ef 17                                       |....|
D (7717) uart-tx: 0x3fcaee40   41 54 2b 49 46 43 3d 32  2c 32 0d                 |AT+IFC=2,2.|
D (7725) uart-tx: 0x3fcaeb98   29 f9                                             |).|
D (7772) uart-rx: 0x3fcc0984   f9 09 ff 0d 0d 0a 4f 4b  0d 0a cd f9              |......OK....|
```

earthPerson-001, 2025-07-16:
If i check from the direct connection with SIM7600 via usb/uart, the response comes in a single line i.e. the +CUSD line. I am not able to understand why the response is separated into two lines only here. If i only search for `corporate` then I can only get the last two words not the whole message.

the main concern is the consistency of the outputs and the reason behind this behavior. I need to determine if this is isolated case or all USSD behave in similar fashion. If there is not consistency then whole system will break if carrier decides to change the response message even the slightest.

earthPerson-001, 2025-07-16:
> I can see from your log that it comes in 4-5 separate CMUX frames.

Is there a reason why only a part of message is separated while other are stitched together perfectly?

I have tried using lambdas as well but i get many other messages in between the expected one and cannot stitch them properly together.

Using UCS i get the following message 
```txt
I (5620) SIM7600: Got phone number:
+CUSD: 0,"Your number is 9802301039, Status: Active, Active date: 21/03/2022, The age of SIM 1214 days, Tariff plan:e�
```

Produced by the code:
```cpp
esp_modem::command_result urc_handler(uint8_t *data, size_t len)
{
    // sim phone number urc
    if (strstr(reinterpret_cast<char *>(data), "+CUSD: 0,\"Your number is") != nullptr) {
        ESP_LOGI("SIM7600", "Got phone number: %s", data);
    }
    return esp_modem::command_result::OK;
}
```

Let me be clear in the uart response. I am checking the raw response by connecting to sim using separate uart terminal. The response with echo on is as follows
```txt
AT+CUSD=1,"*903#",15
OK

+CUSD: 0,"Your number is 9802301039, Status: Active, Active date: 21/03/2022, The age of SIM 1214 days, Tariff plan: Corporate.",15

```

fobos777, 2025-08-02:
I have related problem with SIM7670G using B05 firmware and running in CMUX mode. The reply to  `esp_modem_at_raw(_dce, "AT+CGNSSINFO\r", out, "OK", "ERROR", timeout);` is `\r\nOK\r\n`, but not the expected `+CGNSSINFO: ,,,,,,,,\r\n\r\nOK\r\n`, which I can get in COMMAND mode.