espressif/esp-protocols #898: WEBSOCKET Double free in esp_websocket_client during error/teardown race

filzek, 2025-09-30:
### Answers checklist.

- [x] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [x] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### What component are you using? If you choose Other, provide details in More Information.

esp_websocket_client

### component version

1.5.0

### IDF version.

v5.4.1-18-g6f30fb434d

### More Information.

Hi @glmfe, the websocket double free/tls exception still happening frequently.

Description
When a WebSocket connection aborts unexpectedly, the client sometimes crashes with a TLSF assertion failure indicating a double free. This appears to happen because both the error path and the send/abort path attempt to close and free the same TLS transport, leading to heap corruption.

Observed Behavior

Connection is aborted (errno=113, mbedtls read error :-0x004C).

WEBSOCKET_EVENT_ERROR is fired.

Application code attempts to send or abort, which calls into esp_websocket_client_abort_connection().

The transport is closed twice (esp_transport_close appears twice in the backtrace).

TLS tries to free already-freed memory → assert failed: tlsf_free ... "block already marked as free".

Device reboots with heap corruption.

Backtrace (excerpt)

assert failed: tlsf_free tlsf.c:629 (!block_is_free(block) && "block already marked as free")
esp_tls_conn_destroy -> base_close -> esp_transport_close
ws_close -> esp_transport_close
esp_websocket_client_abort_connection
esp_websocket_client_send_text
EWC_process_queue_tosend
...

Expected Behavior

The WebSocket client should handle error/teardown idempotently.

A single close/free of the TLS transport should occur, even if application code calls abort/stop concurrently with internal error handling.

No heap corruption or double free should be possible.

Steps to Reproduce

Establish a TLS WebSocket connection.

Force the server to close the connection abruptly (RST).

Allow the client task to attempt a send just after the error event is triggered.

Observe crash with TLSF double free assertion.

Environment

ESP-IDF: special build (14.2_20240403 toolchain)
[... 62 lines cut ...]
--- 0x4017ced1: esp_tls_internal_event_tracker_destroy at C:/Espressif/frameworks/esp-idf-special/components/esp-tls/esp_tls_error_capture.c:50
--- 0x4017cc52: esp_tls_conn_destroy at C:/Espressif/frameworks/esp-idf-special/components/esp-tls/esp_tls.c:160
--- 0x4017eba8: base_close at C:/Espressif/frameworks/esp-idf-special/components/tcp_transport/transport_ssl.c:335
--- 0x401fec9d: esp_transport_close at C:/Espressif/frameworks/esp-idf-special/components/tcp_transport/transport.c:172
--- 0x4017f621: ws_close at C:/Espressif/frameworks/esp-idf-special/components/tcp_transport/transport_ws.c:680
--- 0x401fec9d: esp_transport_close at C:/Espressif/frameworks/esp-idf-special/components/tcp_transport/transport.c:172
--- 0x4012f3e4: esp_websocket_client_abort_connection at D:/Dropbox/Dev/wisehome_project/managed_components/espressif__esp_websocket_client/esp_websocket_client.c:240
--- 0x401308bf: esp_websocket_client_send_with_exact_opcode at D:/Dropbox/Dev/wisehome_project/managed_components/espressif__esp_websocket_client/esp_websocket_client.c:662
--- 0x401308e1: esp_websocket_client_send_with_opcode at D:/Dropbox/Dev/wisehome_project/managed_components/espressif__esp_websocket_client/esp_websocket_client.c:1352
--- (inlined by) esp_websocket_client_send_text at D:/Dropbox/Dev/wisehome_project/managed_components/espressif__esp_websocket_client/esp_websocket_client.c:1322
--- 0x40110eeb: EWC_process_queue_tosend at D:/Dropbox/Dev/wisehome_project/main/tasks/websocket/taskWebsocket.c:1121
--- 0x40110f6b: EWC_request_flush at D:/Dropbox/Dev/wisehome_project/main/tasks/websocket/taskWebsocket.c:1244
--- 0x400e650b: WISEON_request_flush at D:/Dropbox/Dev/wisehome_project/main/include/websocket_lib.c:1782
--- 0x4011066f: EWC_enqueue_item_reply at D:/Dropbox/Dev/wisehome_project/main/tasks/websocket/taskWebsocket.c:454
--- (inlined by) EWC_enqueue_item_reply at D:/Dropbox/Dev/wisehome_project/main/tasks/websocket/taskWebsocket.c:369
--- 0x401109b9: WISEON_enqueue_item_reply at D:/Dropbox/Dev/wisehome_project/main/tasks/websocket/taskWebsocket.c:1260
--- 0x400f5038: executar at D:/Dropbox/Dev/wisehome_project/main/include/exec_utils.c:6815
--- 0x401036e4: taskExecCMD at D:/Dropbox/Dev/wisehome_project/main/tasks/taskSentinela.c:4018
--- 0x4009662a: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-special/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
2025-09-30 18:44:11
2025-09-30 18:44:11 Entering gdb stub now.
2025-09-30 18:44:11 $T04#b8GNU gdb (esp-gdb) 14.2_20240403
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.

gabsuren, 2025-10-23:
Hi @filzek , thank you for your detailed testing and reporting these issues!

I've applied a new patch that should resolve both problems:
Please check out the branch below.
 https://github.com/gabsuren/esp-protocols-1/tree/fix/ws_race_on_abort

There's a minor ~400-byte leak still under investigation  (likely in transport layer), but I think it's acceptable for now. 
I'm working on identifying the exact source.

Looking forward to your storm/stress test results tomorrow! 
Best regards,
Suren

filzek, 2025-10-23:
hi @gabsuren, already pull it and send to testing.

also the log printf-format mismatch so i have send a patch to fix it.

We will run the full checkup today in the [ws_race_on_abort](https://github.com/gabsuren/esp-protocols-1/tree/fix/ws_race_on_abort) version today.

gabsuren, 2025-11-05:
@filzek just following up. Could you please let me know if this issue has been solved, either partially or completely?
Also, is there any update or progress you can share?

Thank you!

filzek, 2025-11-17:
Hi @gabsuren 

> @filzek just following up. Could you please let me know if this issue has been solved, either partially or completely?
> Also, is there any update or progress you can share?
> 
> Thank you!

Sorry to take long to answer but we have run the websocket until today and collect all the logs from device and cloud. A total of 180 devices has been tested in more than 30 different places.

There is no much issues in the websoclet now, just some reconnection without justifications at all. But no more double free or memory issues.

Also qe have found that sometimes the 1st packet send form the cloud to the device its get lost, this was traced to some issue inside the websocket as the cloud tracking send the packet, the router track the packet being delivered to the device but the device miss the reception of it, so, something in the websocket is wrong. This just happen to the 1st package and it just randomly.

We have managed to use the 1st reply package to listen to a authcode so, we are sure that no answer is that the package was lost in the device and we resend it and works, so this is a workaround.

We have seen that a new 1.6 version was pushed, does it has all the patches and fixes on i

Did you find out our have saw the missing 1st reply packet ?

That's all.

Thanks

gabsuren, 2025-11-17:
@filzek Glad to hear that things are getting more stable. The new esp_websocket_client v1.6 doesn’t include the fix yet, we wanted to confirm whether it actually improved your failures first. I’ll push it forward so it can be included in the next release.

Regarding the first packet missing issue, I initially thought that this PR would resolve it: https://github.com/espressif/esp-protocols/pull/924
 — but it seems it didn’t. I’ll investigate it further.

Thank you!

gabsuren, 2025-12-25:
Hi @khbenjamin - are you using the latest master of the esp-protocols WebSocket client? I pushed a fix to master last week. Could you please check if your version includes this https://github.com/espressif/esp-protocols/pull/924/commits/23ca97d5eccff866e89cb880df3e8366bf7d12b7

gabsuren, 2025-12-25:
@khbenjamin Please note that the fix is currently only available in master and has not yet been included in a released version of the websocket component yet.

gabsuren, 2025-12-26:
Closing this issue as the fix has already been merged into `master`. Please reopen it if you’re still experiencing the problem.

Regards,
Suren

closed 2025-12-26 by gabsuren

PoohWoah, 2025-12-29:
> > 好[@khbenjamin](https://github.com/khbenjamin)您使用的是 esp-protocols WebSocket 客户端的最新主分支吗？我上周向主分支进行了一个修复程序。请问您的版本是否包含[23ca97d更新你？](https://github.com/espressif/esp-protocols/commit/23ca97d5eccff866e89cb880df3e8366bf7d12b7)
> 
> 您好，我目前正在使用最新版本的 esp_websocket_client 库，但仍然遇到超时和连接断开的问题。以下是我的日志： `E (751175) websocket_client: Could not lock ws-client within 2000 timeout for PONG E (751175) websocket_client: Error receive data E (751178) transport_base: poll_write select error 0, errno = Success, fd = 52 E (751185) transport_ws: Error transport_poll_write app_ws: WebSocket disconnected...`
> 
> E (751194) websocket_client: esp_transport_write() 返回 -1，transport_error=ESP_OK，tls_error_code=0，tls_flags=0，errno=0 app_ws: Websocket --错误！！！！！！
> 
> app_ws：Webscoket 已连接...

network_timeout_ms is 1000ms

PoohWoah, 2025-12-30:
Unfortunately, after a full day of testing yesterday, I still encountered the following problem:
`E (60519122) transport_ws: Error read data
E (60519122) transport_ws: Error reading payload data
app_ws: ws data->fin: 1, data->op_code: 1
app_ws: data_len: 0, payload_len: 1157, payload_offset: 954
app_ws: data_len: 954, payload_len: 1157, payload_offset: 0
app_w�`���̗�Kx�>�/�*G�)� a�6�,�K4o{��Z�?`��m    ]�<I�-�J���/��V֋'~Ep�y��lz��_O�`�\X�o�`��z?���^Y!X�1O��6��%3��������y{��Z`��^�r����:��գ'C��v�����W8<�w��Vɇ�y��
��6L�   w�}�7/�>
app_w�`���̗�Kx�>�/�*G�)� a�6�,�K4o{��Z�?`��m    ]�<I�-�J���/��V֋'~Ep�y��lz��_O�`�\X�o�`��z?���^Y!X�1O��6��%3��������y{��Z`��^�r����:��գ'C��v�����W8<�w��Vɇ�y��
Z       w�}�7/�>
app_ws: moonraker_ws_rcv_decode data[103]:
e": 11.3, "memory": 24348, "mem_units": "kB"}, "cpu_temp": 39.329, "network": {"lo": {"rx_bytes": 3966,
app_ws: Websocket closed...`